Part-DB/Part-DB-server #2: General State of Development

rs-development, 2019-10-12:
I see you are working on the project right now. First big thanks for all the effort you are putting into that project!.

I would like to ask if you could share a quick overview of what features are tested and which are missing or not working. And in how far the databases are compatible with the older version.

We start a new project space and are thinking about using the new version...

jbtronics, 2019-10-13:
Here some overview what is working in the current version and what features are missing in the moment in comparision with Part-DB 0.6

### Working:
- Basic part managment (creating, viewing, editing, deleting parts)
- Managments of datastructures (categories, storelocations, etc...) (create, editing, delete, (most likely) import/export of them)
- Basic part tables (showing parts in a category, storelocation, etc.)
- User system and the almost all parts of the permission system
- Attachments
- Editing/viewing price informations (with currency and conversion abilities)
- Migration from old Part-DB versions

### Missing yet:
- Device/projects
- Label generator and possibility to quick access parts via label
- Advanced search (you can not specify in which fields should be searched), no live search
- Order system (things like marking parts for ordering and list them later)
- Mass editing parts via tables
- 3D-Footprints
- Parts property table created from description/comment
- Tools like Calculator or SMD Labels (I am not sure yet if i will put them in the new version at all)
- Import/Export of parts
- Event log (so no part history, login history, etc.)
- Statistics view
- No generation of property tables out of part description
- Automatic created links to datasheets
- System and database settings page

You can see that the basic inventory managment features are working, but the new version is not really usage tested yet and sometimes a bit unpolished.

It is possible to migrate an database of Part-DB 0.5/0.6 to use with the new version, but after that it is not possible to access the database from the old versions any more.

Overall I would not  recommend Part-DB 1.0 in the current version for usage in production, unless you really need the new features (like multiple storelocations and tags for parts) and dont need the missing features.
Once Part-DB 1.0 is more feature complete you can upgrade from the old version easily, so there should be no disadvantage if you start your new project with the older Part-DB versions (Part-DB 0.5.7 for example)

rs-development, 2019-10-14:
Thanks for the effort you put into this! We will work with version 0.6 for now. 

Deeper into the winter, if our people start using it, I will try and help with some functionalities!

closed 2019-10-14 by rs-development