Part-DB/Part-DB-server #3: db schema creation failure

alexanderustinov, 2019-12-24:
Greetings.
Trying to make clean install of this rewrite of part-db.
Execution of `php bin/console doctrine:migrations:migrate` breaks with following:
```
In AbstractMySQLDriver.php line 79:
                                                                             
  An exception occurred while executing 'UPDATE `groups`                     
          SET perms_parts_category = 9, perms_parts_minamount = 9, perms_pa  
  rts_lots = 169, perms_parts_tags= 9,                                       
              perms_parts_unit = 9, perms_parts_mass = 9, perms_parts_statu  
  s = 9, perms_parts_mpn = 9,                                                
              perms_currencies = 9897, perms_measurement_units = 9897, perm  
  s_parts_attachements = 681,                                                
              perms_parts_orderdetails = 681, perms_parts_prices = 681       
          WHERE id = 2 AND name = "readonly";                                
                                                                             
          UPDATE `groups`                                                    
          SET perms_parts_category = 5, perms_parts_minamount = 5, perms_pa  
  rts_lots = 85, perms_parts_tags= 5,                                        
              perms_parts_unit = 5, perms_parts_mass = 5, perms_parts_statu  
  s = 5, perms_parts_mpn = 5,                                                
              perms_currencies = 5461, perms_measurement_units = 5461, perm  
  s_parts_attachements = 341,                                                
              perms_parts_orderdetails = 341, perms_parts_prices = 341       
          WHERE (id = 1 AND name = "admins")                                 
              OR (id = 3 AND name = "users");                                
          ':                                                                 
                                                                             
  You have an error in your SQL syntax; check the manual that corresponds t  
  o your MariaDB server version for the right syntax to use near 'UPDATE `g  
  roups`                                                                     
          SET perms_parts_category = 5, perms_parts_minamount = 5,' at line  
   8                                                                         
                                                                             

In MysqliStatement.php line 82:
                                                                             
  You have an error in your SQL syntax; check the manual that corresponds t  
  o your MariaDB server version for the right syntax to use near 'UPDATE `g  
  roups`                                                                     
          SET perms_parts_category = 5, perms_parts_minamount = 5,' at line  
   8                                                                                                                                                                                                                                                          
```
I'm on Gentoo with php 7.3.11 and mariadb 10.2.29.

jbtronics, 2019-12-24:
Interesting. I guess that your MariaDB Version (I use MariaDB 10.4 for development, and it does not show this problem) has the problem with that line, because it has two commands in a single SQL query.

I have split this command into two different SQL queries and I think it should work now.
I would be grateful, if you could test this (remove all tables from your DB before retrying the migration)

alexanderustinov, 2019-12-24:
Fixed. Thanks.

closed 2019-12-24 by alexanderustinov