Part-DB/Part-DB-server #5: I must be missing something major

a7hybnj2, 2020-01-01:
I cannot find in the DEMO SITE where to add a new part. I would think this is like a MAIN feature and I must be blind to not be able to find it.

In the non-symfony version there is a 'new part in this category' button that I cannot seem to find the equivalent of in this version.

jbtronics, 2020-01-01:
The part lists have no part create button yet, because I have not thought about how to pass the information, which list you had viewed to the part creation dialog (especially when we have some more complex filters).

In the moment you can use the "Edit / Part" entry of the tool menu (I will rename that to new part later) or when you are on the page of an existing part, you can use the drop-down menu next to the "clone part" button, which also gives you an empty part creation dialog.

a7hybnj2, 2020-01-02:
Thanks for the information! Part-DB is great so I am really looking forward to the updated symfony version!

jbtronics, 2020-04-01:
I have added buttons to quickly create parts in the given category from a part list.

closed 2020-04-01 by jbtronics