Part-DB/Part-DB-server #19: Check for argon2 in php?

alexanderustinov, 2020-01-30:
It taken me few days on gentoo to get a grip on silent 500 error due to php compiled without '--with-password-argon2' flag.

Is it straightforward to check for that  with composer?
Adding it to composer.json like
```json
        "php": "^7.2.5",
        "ext-bcmath": "*",
        "ext-ctype": "*",
        "ext-gd": "*",
        "ext-iconv": "*",
        "ext-intl": "*",
        "ext-json": "*",
        "ext-mbstring": "*",
        "ext-password-argon2": "*"
```
doesn't work.

jbtronics, 2020-01-30:
What shows up in the error log file?

Part-DB should not require argon2. If it is available Symfony will use it for password encryption (it is the strongest one available), but if not available bcrypt will be used.

Have you importet an database from a server which has used argon2? (In that case Part-DB would not be able to validate the argon2 password hashes).

alexanderustinov, 2020-01-30:
Server gave 500 in reply to login (with initial admin/admin credentials).

Apache log contain nothing, php-fpm log also nothing.
(when i introduce error to index.php by hands it gets to the logs)
Finally, in dev mode i spotted
```
[2020-01-30 17:04:47] doctrine.DEBUG: UPDATE `users` SET password = ?, last_modified = ? [2020-01-30 17:04:46] php.INFO: User Deprecated: Calling Doctrine\ORM\EntityManager::flush() with any arguments to flush specific entities is deprecated and will not be supported in Doctrin>
[2020-01-30 17:04:47] doctrine.DEBUG: "START TRANSACTION" [] []
[2020-01-30 17:04:47] doctrine.DEBUG: UPDATE `users` SET password = ?, last_modified = ? WHERE id = ? ["$argon2id$v=19$m=65536,t=4 [...]","2020-01-30 17:04:47",2] []
[2020-01-30 17:04:47] doctrine.DEBUG: "COMMIT" [] []
```
in the application log. Which lead to solution.
I'm trying to do fresh setup, DB was not imported. I use mariadb-10.2.29.

alexanderustinov, 2020-01-30:
It should be upstreamed to php?

jbtronics, 2020-01-30:
The output does not look like missing argon caused the error (if it were missing, it would be failing much earlier while generating the hash. The DB log clearly shows the Argon password hash header, so that hash generation looks fine).

Could you send me the var/log/prod.log (or dev.log) from your DB error. If an error happened a stack trace can be found there, which helps much to find the bug.

jbtronics, 2020-01-30:
Has you run php bin/console doctrine:migrations:migrate and it showed no errors?

alexanderustinov, 2020-01-30:
> Has you run php bin/console doctrine:migrations:migrate and it showed no errors?
Yes, no errors. Front page also worked. Everything was fine until login.

Will try to break it back now and capture the logs.

alexanderustinov, 2020-01-31:
"Measurements" results:
1. Starting fresh app build with argon2 disabled in php.
1. `composer install` (both --no-dev and without argument) `&& yarn install && yarn build` gives 500 error. Nothing in any (apache, php-fpm, app) log. With both "prod" and "dev". About app log i am not sure, as there could have been permission issues.
1. After php rebuilt with argon2 enabled: "dev" version "throws" about inability to write log. After `chmod 777 var/log` it works. "prod" version also just works as expected.
1. Rebuilt php once again disabling argon2, on login app states invalid credentials with following in the logs
```
[2020-01-31 16:28:46] security.INFO: Authentication request failed. {"exception":"[object] (Symfony\\Component\\Security\\Core\\Exception\\BadCredentialsException(code: 0): Bad credentials. at /home/builder/partdb/vendor/symfony/security-core/Authentication/Provider/UserAuthenticationProvider.php:85, Symfony\\Component\\Security\\Core\\Exception\\BadCredentialsException(code: 0): The presented password is invalid. at /home/builder/partdb/vendor/symfony/security-core/Authentication/Provider/DaoAuthenticationProvider.php:65)"} []
[2020-01-31 16:28:46] security.DEBUG: Authentication failure, redirect triggered. {"failure_path":"login"} []
[2020-01-31 16:28:46] request.INFO: Matched route "login". {"route":"login","route_parameters":{"_route":"login","_controller":"App\\Controller\\SecurityController::login","_locale":"en"},"request_uri":"http://11.0.0.102:81/en/login","method":"GET"} []
```
5. Yet again rebuild php with argon2 enabled. Works. 

That is a nasty one :)

closed 2020-04-08 by jbtronics