Part-DB/Part-DB-server #20: console cache:warmup silently fails when no permission to write to var/cache

alexanderustinov, 2020-01-31:
`php bin/console cache:warmup` script mentioned in [installation notes](https://github.com/Part-DB/Part-DB-symfony#installation) states succesfull warming, but logs errors. Trying to open app in browser after that gives 500 error code.
```
2020-01-31 17:17:44] cache.WARNING: Failed to save key "version" of type Version\Version: file_put_contents(/home/builder/partdb/var/cache/prod/pools/OqCSUa9WTD/5e346138abd444.61684305): failed to open stream: Permission denied {"key":"version","exception":"[object] (ErrorException(code: 0): file_put_contents(/home/builder/partdb/var/cache/prod/pools/OqCSUa9WTD/5e346138abd444.61684305): failed to open stream: Permission denied at /home/builder/partdb/vendor/symfony/cache/Traits/FilesystemCommonTrait.php:98)"} []
[2020-01-31 17:17:44] security.DEBUG: Stored the security token in the session. {"key":"_security_main"} []
[2020-01-31 17:17:45] request.INFO: Matched route "user_settings". {"route":"user_settings","route_parameters":{"_route":"user_settings","_controller":"App\\Controller\\UserSettingsController::userSettings","_locale":"en"},"request_uri":"http://11.0.0.102:81/en/user/settings","method":"GET"} []
[2020-01-31 17:17:45] security.DEBUG: Read existing security token from the session. {"key":"_security_main","token_class":"Symfony\\Component\\Security\\Core\\Authentication\\Token\\UsernamePasswordToken"} []
[2020-01-31 17:17:45] security.DEBUG: User was reloaded from a user provider. {"provider":"Symfony\\Bridge\\Doctrine\\Security\\User\\EntityUserProvider","username":"admin"} []
[2020-01-31 17:17:45] request.CRITICAL: Uncaught PHP Exception Symfony\Component\Filesystem\Exception\IOException: "Unable to write to the "/home/builder/partdb/var/cache/prod" directory." at /home/builder/partdb/vendor/symfony/filesystem/Filesystem.php line 683 {"exception":"[object] (Symfony\\Component\\Filesystem\\Exception\\IOException(code: 0): Unable to write to the \"/home/builder/partdb/var/cache/prod\" directory. at /home/builder/partdb/vendor/symfony/filesystem/Filesystem.php:683)"} []
[2020-01-31 17:17:45] cache.WARNING: Failed to save key "%5BSymfony%5CComponent%5CHttpKernel%5CController%5CErrorController%5D%5B1%5D" of type array: file_put_contents(/home/builder/partdb/var/cache/prod/pools/yt4qg5foDg/5e3461392b71d7.69584045): failed to open stream: Permission denied {"key":"%5BSymfony%5CComponent%5CHttpKernel%5CController%5CErrorController%5D%5B1%5D","exception":"[object] (ErrorException(code: 0): file_put_contents(/home/builder/partdb/var/cache/prod/pools/yt4qg5foDg/5e3461392b71d7.69584045): failed to open stream: Permission denied at /home/builder/partdb/vendor/symfony/cache/Traits/FilesystemCommonTrait.php:98)"} []
[2020-01-31 17:17:45] cache.WARNING: Failed to save key "%5BSymfony%5CComponent%5CHttpKernel%5CController%5CErrorController%23__invoke%5D%5B1%5D" of type array: file_put_contents(/home/builder/partdb/var/cache/prod/pools/yt4qg5foDg/5e3461392b71d7.69584045): failed to open stream: Permission denied {"key":"%5BSymfony%5CComponent%5CHttpKernel%5CController%5CErrorController%23__invoke%5D%5B1%5D","exception":"[object] (ErrorException(code: 0): file_put_contents(/home/builder/partdb/var/cache/prod/pools/yt4qg5foDg/5e3461392b71d7.69584045): failed to open stream: Permission denied at /home/builder/partdb/vendor/symfony/cache/Traits/FilesystemCommonTrait.php:98)"} []
[2020-01-31 17:17:45] cache.WARNING: Failed to save key "version" of type Version\Version: file_put_contents(/home/builder/partdb/var/cache/prod/pools/OqCSUa9WTD/5e3461392bfe08.42085575): failed to open stream: Permission denied {"key":"version","exception":"[object] (ErrorException(code: 0): file_put_contents(/home/builder/partdb/var/cache/prod/pools/OqCSUa9WTD/5e3461392bfe08.42085575): failed to open stream: Permission denied at /home/builder/partdb/vendor/symfony/cache/Traits/FilesystemCommonTrait.php:98)"} []
[2020-01-31 17:17:45] security.DEBUG: Stored the security token in the session. {"key":"_security_main"} []
```

jbtronics, 2020-01-31:
This looks like some weird file permission bug. Does the web-server user, has write access to the var/ folder? You should maybe do an `chown -R www-data:www-data var/*` in your part-db installation folder. You could also try to delete the var/cache/ folder completley, so the cache is created fresh.

I would guess that this is somehow related to your other issue. (If symfony has cached that it can use Argon, this could explain your error). Also if you change something on your PHP you should do a php bin/console cache:clear, so the cache is regenerated.

alexanderustinov, 2020-01-31:
Chmod helped, yes. But the point was in cache:warmup not raising an exception in case of failure.

The other issue was debugged starting with freshly cloned repo and php was without argon at first build of application. Suppose, something in auth code or its dependencies needs argon to function.

closed 2020-04-08 by jbtronics