Part-DB/Part-DB-server #23: path prefix from webpack.config.js not used

sjoerdtimmer, 2020-02-28:
I have configured the desired path prefix as described in the installation notes:
```
   .setPublicPath('/part-db-symfony/build')
    .setManifestKeyPrefix('build/')
```
When I visit http://myserver/part-db-symfony/index.php I get redirected to http://myserver/en/part-db-symfony/index.php (i.e. there is an extra /en/ prefixed before the path).

Any suggestions about what I may be doing wrong or if this is a real bug?

jbtronics, 2020-02-29:
This has nothing to do with the webpack config. I guess that this is related to redirecting that happens when no locale is specified and that Symfony can not determine the correct base path.

What happens if you visit `http://myserver/part-db-symfony/index.php/en` directly?
What happens if you set `NO_URL_REWRITE_AVAILABLE` to 1 in your `.env.local` (or `.env`) file (NO_URL_REWRITE_AVAILABLE=1)?

How does the nginx configuration for the Part-DB folder looks like? (It seems that you did not have setup url rewriting).

sjoerdtimmer, 2020-03-03:
I was really missing a lot here. My server was not configured to process anything through php if the url does not end in \.php$ 

I added `NO_URL_REWRITE_AVAILABLE=1` and re-ran `composer install --no-dev` but `/part-db-symfony/index.php` still redirects to `/en/part-db-symfony/index.php`.

I was trying to install part-db-symfony alongside part-db on a host that I need to access using ssh port forwarding (i.e. no SNI). I'm currently successfully running them side by side on different ports (don't know why I didn't think of that before).

Even though the install in a sub-directory never worked for me despite following the instructions from the wiki and the symfony website I'm closing the issue now.

Thanks for your support!

closed 2020-03-03 by sjoerdtimmer