Part-DB/Part-DB-server #26: Letzte Aktivitäten nicht aussagekräftig

coeka, 2020-03-09:
Ich habe die Beschreibung des Lagerorts geändert:
[image]

Im Log auf der Startseite steht dann:
[image]

1. Ich denke, da sollte dann auch ersichtlich sein, welches Bauteil ich geändert habe.
2. Als geänderte Eigenschaft ist description nicht eindeutig, es könnte genauso gut die Bauteil-Beschreibung gemeint sein und nicht die vom Lagerort.
3. Ich habe nicht den Preis geändert, steht trotzdem im Log

coeka, 2020-03-15:
Das passiert nicht nur beim Preis, dass der als geänderte Eigenschaft im Log steht, obwohl nicht geändert. Hier habe ich zum Beispiel nur die Beschreibung geändert, im Log steht aber das ich auch das Expiration Date geändert habe

[image]

jbtronics, 2020-03-29:
Sollte behoben sein.

closed 2020-03-29 by jbtronics