Part-DB/Part-DB-server #36: "Create new Part" button on "not selectable" categories

tbnobody, 2020-04-06:
**Describe the bug**
When selecting a category in the category list a button called "create new part" appears. But this button also appears if the category is marked as "not selectable". (also occurs in the demo system)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Categories and choose a category which is marked as "not selectable".
2. Press the "Create new part" button (which should maybe not be visible)
3. The "not selectable" category is pre-selected (which should not be possible)

**Expected behavior**
Hide the button on "not selectable" categories.

**Server Side**
- Part-DB Version: e654c46e11d5eab780c12f069a4d2d2050ae2b65
- PHP Version: PHP 7.3.16
- Database Server 10.3.22-MariaDB-0+deb10u1 Debian 10

**Desktop:**
 - OS: Windows 10
 - Browser Vivaldi
 - Version 2.11.1811.52 (Stable channel) (32-Bit)

**Additional context**
I am not sure whether you already like such error reports for the dev version?

jbtronics, 2020-04-06:
Thank you for your bug report. I'm always happy to get feedback.

I have fixed it: Now the button is disabled (grayed out) when the category is marked as "not selectable"

closed 2020-04-06 by jbtronics

tbnobody, 2020-04-06:
Hm note sure whether I missed something during the update, but the button is always grey now.

tbnobody, 2020-04-06:
After looking a bit in the code, is it possible, that the property is called `isNotSelectable` and not `notSelectable` ?

tbnobody, 2020-04-06:
oh sorry... my fault... session was expired, I wasn't logged in... --> works perfectly :)