Part-DB/Part-DB-server #39: Unable to move a footprint to a category which is configured as "not selectable"

tbnobody, 2020-04-06:
**Describe the bug**
It is only possible to move a footprint to a category which is NOT configured as "not selectable". During the creation of a footprint all parents can be selected, but when editing the same footprint it's not possible to select footprints configured as "not selectable".

**To Reproduce**
Steps to reproduce the behavior:
1. Create a footprint and call it "a" --> configure as "not selectable" --> Save
2. Create a footprint and call it "b" --> don't choose any parent --> Save
3. Open footprint "b" again and try to assign it to parent "a"

**Expected behavior**
Do I understand the "not selectable" flag wrong? I assumed it should only apply when selecting parts and not for e.g. moving one footprint from one parent to another.

**Server Side**
- Part-DB Version: e654c46e11d5eab780c12f069a4d2d2050ae2b65
- PHP Version: PHP 7.3.16
- Database Server 10.3.22-MariaDB-0+deb10u1 Debian 10

**Desktop:**
 - OS: Windows 10
 - Browser Vivaldi
 - Version 2.11.1811.52 (Stable channel) (32-Bit)

**Additional context**
I am not sure whether you already like such error reports for the dev version?

jbtronics, 2020-04-08:
What exactly happens, if you try to assign a as parent element of b? Is the option of a greyed out?

I can not reproduce the bug...

tbnobody, 2020-04-08:
It's a bit tricky to reproduce. I just tried it and it seem to work. But the problem seems to be where a is located:
[image]

As you can see in this screenshot I am unable to assign 01005 to test (greyed out)

jbtronics, 2020-04-08:
Interesting.

Have you tried to clean the cache, by executing "php bin/console cache:clear" in a console?

tbnobody, 2020-04-08:
I have just cleared the cache as you described above, but I see the same result.

But I reproduced it now! It only occurs for footprints with an attachment!

Please see: https://part-db.herokuapp.com/en/footprint/74/edit#common
and try to move it to a, b or c

jbtronics, 2020-04-09:
This is fixed now.

closed 2020-04-09 by jbtronics