Part-DB/Part-DB-server #45: Mass-Creation of Parts

MilanDeruelle, 2020-04-12:
**Is your feature request related to a problem? Please describe.**
When importing multiple similar parts (e.g. I have a box of trim potis - every property is the same, except the resistance). Adding them all, requires a lot of manual repetitive work.

**Describe the solution you'd like**
Allow for the mass-creation, e.g. by selecting a base-item, provide a "more like this" button and then proving a CSV field to overwrite properties of the base-item (maybe using placeholders in the base-item?)

**Describe alternatives you've considered**
A simple fix could consist of a save & create more like this button allowing to create a second copy of a created part, so only the changed property has to be changed on the new part.

coeka, 2020-04-17:
You can already clone a part in the current version, so you don't have to set every property for a new part.
I know, this is not a mass-creation, i just want to make sure, that you know this feature exists and can help you a little bit.

MilanDeruelle, 2020-04-17:
Hi & thanks for the answer. How do I use cloning? This is indeed at least a help, even tough creating a whole bunch of items, e.g. for a resistor series is still a feature I'd like to see. I have duplicate buttons for devices, locations, etc. but using the "new part" button I only get the option to save.

jbtronics, 2020-05-30:
In the dropdown next to the save button there is now the option "save and clone", which save the current part, and let you create a new part with the other one as base.

For now I will closing this issue, if you have other ideas how to improve the part creation workflow, I guess it would be best to open a new issue.

closed 2020-05-30 by jbtronics