Part-DB/Part-DB-server #59: Error 500 when adding stock to part

tbnobody, 2020-05-17:
**Describe the bug**
I created a part, saved it and tried to add stock. I got the following error message:
`[2020-05-17T14:51:05.730722+02:00] request.CRITICAL: Uncaught PHP Exception Doctrine\ORM\ORMInvalidArgumentException: "Binding entities to query parameters only allowed for entities that have an identifier." at /var/www/partdb/vendor/doctrine/orm/lib/Doctrine/ORM/ORMInvalidArgumentException.php line 207 {"exception":"[object] (Doctrine\\ORM\\ORMInvalidArgumentException(code: 0): Binding entities to query parameters only allowed for entities that have an identifier. at /var/www/partdb/vendor/doctrine/orm/lib/Doctrine/ORM/ORMInvalidArgumentException.php:207)"} []`

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://part-db.herokuapp.com/en/part/970/edit#part_lots
2. Click on Add stock
3. Select a storage location, enter any amount
4. See error

**Server Side**
- Part-DB Version: cad9534b04a231e2ad2ed87c13f9b3f8ddd1bbad
- PHP Version: PHP 7.3.16
- Database Server 10.3.22-MariaDB-0+deb10u1 Debian 10

**Desktop:**
 - OS: Windows 10
 - Browser Firefox
 - Version 76.0.1 (64-Bit)

tbnobody, 2020-05-17:
Additional remark, I get the same error message when I try to add a supplier to the part.

jbtronics, 2020-05-17:
fixed.

closed 2020-05-17 by jbtronics