Part-DB/Part-DB-server #64: Save column settings per type and user

tbnobody, 2020-06-01:
**Is your feature request related to a problem? Please describe.**
When e.g. looking at the list of categories i don't wanna see the detailed description. It's now possible to hide this column but this setting is not saved and therefor I have to hide it every time I click on a category.

**Describe the solution you'd like**
It would be nice to save this column settings (hidden/shown, sorting direction) for every user and every type of list (e.g. different settings for categories and store locations etc.)

virtadpt, 2022-08-03:
This would be really nice to have.

jbtronics, 2022-09-12:
This is now partially implemented, as the column visibility is now saved locally across all part tables, so you do not have to change the column visibility any time you visit another page anymore.
This setting is saved locally in the browser, meaning it is not synced across different devices (and not really saved per user). This would be difficult however, especially if you the same user is logged in on two devices (and select different column visibility there)

influx3k, 2023-11-14:
Still resets on the same machine and same browser after some period of time. As little as a 2-3 hours.