Part-DB/Part-DB-server #67: database upgrade from old part-db

coeka, 2020-06-05:
**Describe the bug**
I tried an upgrade from the old part-db-version, but it doesn't work. At some time i get an error message, see below. I also tried to drop the database and restore the backup, but it also didn't work.

**Server Side**
- Part-DB Version: Part-DB 1.0 
- PHP Version: 7.3
- Database Server 10.3.22-MariaDB-0+deb10u1 Raspbian 10

**Additional context**

pi@raspberrypi:/var/www/part-db $ php bin/console doctrine:migrations:migrate

                    Application Migrations

WARNING! You are about to execute a database migration that could result in schema changes and data loss. Are you sure you wish to continue? (y/n)y
Migrating up to 20200502161750 from 0

  ++ migrating 1 (Creates an inital empty database)

  SS skipped (Reason: Old Part-DB Database detected! Continue with upgrade...)

  ++ migrating 20190902140506 (Upgrade database from old Part-DB 0.5 Version (dbVersion 26))

     -> SET sql_mode = ''
     -> RENAME TABLE `attachement_types` TO `attachment_types`;
     -> RENAME TABLE `attachements` TO `attachments`;
     -> CREATE TABLE currencies (id INT AUTO_INCREMENT NOT NULL, parent_id INT DEFAULT NULL, iso_code VARCHAR(255) NOT NULL, exchange_rate NUMERIC(11, 5) DEFAULT NULL, comment LONGTEXT NOT NULL, not_selectable TINYINT(1) NOT NULL, name VARCHAR(255) NOT NULL, last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, datetime_added DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, INDEX IDX_37C44693727ACA70 (parent_id), PRIMARY KEY(id)) DEFAULT CHARACTER SET utf8mb4 COLLATE utf8mb4_unicode_ci ENGINE = InnoDB
     -> CREATE TABLE `measurement_units` (id INT AUTO_INCREMENT NOT NULL, parent_id INT DEFAULT NULL, unit VARCHAR(255) DEFAULT NULL, is_integer TINYINT(1) NOT NULL, use_si_prefix TINYINT(1) NOT NULL, comment LONGTEXT NOT NULL, not_selectable TINYINT(1) NOT NULL, name VARCHAR(255) NOT NULL, last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, datetime_added DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, INDEX IDX_F5AF83CF727ACA70 (parent_id), PRIMARY KEY(id)) DEFAULT CHARACTER SET utf8mb4 COLLATE utf8mb4_unicode_ci ENGINE = InnoDB
     -> CREATE TABLE part_lots (id INT AUTO_INCREMENT NOT NULL, id_store_location INT DEFAULT NULL, id_part INT NOT NULL, description LONGTEXT NOT NULL, comment LONGTEXT NOT NULL, expiration_date DATETIME DEFAULT NULL, instock_unknown TINYINT(1) NOT NULL, amount DOUBLE PRECISION NOT NULL, needs_refill TINYINT(1) NOT NULL, last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, datetime_added DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, INDEX IDX_EBC8F9435D8F4B37 (id_store_location), INDEX IDX_EBC8F943C22F6CC4 (id_part), PRIMARY KEY(id)) DEFAULT CHARACTER SET utf8mb4 COLLATE utf8mb4_unicode_ci ENGINE = InnoDB
     -> INSERT IGNORE INTO part_lots (id_part, id_store_location, amount, instock_unknown, last_modified, datetime_added) SELECT parts.id, parts.id_storelocation,  parts.instock, 0, NOW(), NOW() FROM parts WHERE parts.instock >= 0
     -> INSERT IGNORE INTO part_lots (id_part, id_store_location, amount, instock_unknown, last_modified, datetime_added) SELECT parts.id, parts.id_storelocation, 0, 1, NOW(), NOW() FROM parts WHERE parts.instock = -2
     -> ALTER TABLE currencies ADD CONSTRAINT FK_37C44693727ACA70 FOREIGN KEY (parent_id) REFERENCES currencies (id)
     -> ALTER TABLE `measurement_units` ADD CONSTRAINT FK_F5AF83CF727ACA70 FOREIGN KEY (parent_id) REFERENCES `measurement_units` (id)
     -> ALTER TABLE part_lots ADD CONSTRAINT FK_EBC8F9435D8F4B37 FOREIGN KEY (id_store_location) REFERENCES `storelocations` (id)
     -> ALTER TABLE part_lots ADD CONSTRAINT FK_EBC8F943C22F6CC4 FOREIGN KEY (id_part) REFERENCES `parts` (id) ON DELETE CASCADE
     -> ALTER TABLE parts DROP INDEX parts_order_orderdetails_id_k, ADD UNIQUE INDEX UNIQ_6940A7FE81081E9B (order_orderdetails_id)
     -> ALTER TABLE parts DROP FOREIGN KEY parts_id_storelocation_fk
     -> DROP INDEX favorite ON parts
     -> DROP INDEX parts_id_storelocation_k ON parts
     -> ALTER TABLE parts DROP FOREIGN KEY parts_id_footprint_fk
     -> ALTER TABLE parts DROP FOREIGN KEY parts_id_manufacturer_fk
     -> ALTER TABLE parts CHANGE mininstock minamount DOUBLE PRECISION NOT NULL, ADD id_part_unit INT DEFAULT NULL, ADD manufacturer_product_number VARCHAR(255) NOT NULL, ADD manufacturing_status VARCHAR(255) DEFAULT NULL, ADD needs_review TINYINT(1) NOT NULL, ADD tags LONGTEXT NOT NULL, ADD mass DOUBLE PRECISION DEFAULT NULL, DROP instock, CHANGE id_category id_category INT NOT NULL, CHANGE name name VARCHAR(255) NOT NULL, CHANGE order_quantity order_quantity INT NOT NULL, CHANGE manual_order manual_order TINYINT(1) NOT NULL, CHANGE manufacturer_product_url manufacturer_product_url VARCHAR(255) NOT NULL, CHANGE last_modified last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, CHANGE favorite favorite TINYINT(1) NOT NULL, DROP id_storelocation
     -> ALTER TABLE parts ADD CONSTRAINT FK_6940A7FE5697F554 FOREIGN KEY (id_category) REFERENCES `categories` (id)
     -> ALTER TABLE parts ADD CONSTRAINT FK_6940A7FEEBBCC786 FOREIGN KEY (id_master_picture_attachement) REFERENCES `attachments` (id)
     -> ALTER TABLE parts ADD CONSTRAINT FK_6940A7FE2626CEF9 FOREIGN KEY (id_part_unit) REFERENCES `measurement_units` (id)
     -> CREATE INDEX IDX_6940A7FEEBBCC786 ON parts (id_master_picture_attachement)
     -> CREATE INDEX IDX_6940A7FE2626CEF9 ON parts (id_part_unit)
     -> CREATE INDEX IDX_6940A7FE5697F554 ON parts (id_category)
     -> DROP INDEX parts_id_category_k ON parts
     -> DROP INDEX parts_id_footprint_k ON parts
     -> CREATE INDEX IDX_6940A7FE7E371A10 ON parts (id_footprint)
     -> DROP INDEX parts_id_manufacturer_k ON parts
     -> CREATE INDEX IDX_6940A7FE1ECB93AE ON parts (id_manufacturer)
     -> ALTER TABLE parts ADD CONSTRAINT parts_id_footprint_fk FOREIGN KEY (id_footprint) REFERENCES footprints (id)
     -> ALTER TABLE parts ADD CONSTRAINT parts_id_manufacturer_fk FOREIGN KEY (id_manufacturer) REFERENCES manufacturers (id)
     -> ALTER TABLE attachment_types DROP FOREIGN KEY attachement_types_parent_id_fk
     -> ALTER TABLE attachment_types ADD filetype_filter LONGTEXT NOT NULL, ADD not_selectable TINYINT(1) NOT NULL, CHANGE name name VARCHAR(255) NOT NULL, CHANGE comment comment LONGTEXT NOT NULL, CHANGE last_modified last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL
     -> DROP INDEX attachement_types_parent_id_k ON attachment_types
     -> CREATE INDEX IDX_EFAED719727ACA70 ON attachment_types (parent_id)
     -> ALTER TABLE attachment_types ADD CONSTRAINT attachement_types_parent_id_fk FOREIGN KEY (parent_id) REFERENCES attachment_types (id)
     -> ALTER TABLE categories DROP FOREIGN KEY categories_parent_id_fk
     -> ALTER TABLE categories ADD not_selectable TINYINT(1) NOT NULL, CHANGE name name VARCHAR(255) NOT NULL, CHANGE disable_footprints disable_footprints TINYINT(1) NOT NULL, CHANGE disable_manufacturers disable_manufacturers TINYINT(1) NOT NULL, CHANGE disable_autodatasheets disable_autodatasheets TINYINT(1) NOT NULL, CHANGE disable_properties disable_properties TINYINT(1) NOT NULL, CHANGE comment comment LONGTEXT NOT NULL, CHANGE last_modified last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL
     -> DROP INDEX categories_parent_id_k ON categories
     -> CREATE INDEX IDX_3AF34668727ACA70 ON categories (parent_id)
     -> ALTER TABLE categories ADD CONSTRAINT categories_parent_id_fk FOREIGN KEY (parent_id) REFERENCES categories (id)
     -> ALTER TABLE devices DROP FOREIGN KEY devices_parent_id_fk
     -> ALTER TABLE devices ADD not_selectable TINYINT(1) NOT NULL, CHANGE name name VARCHAR(255) NOT NULL, CHANGE order_quantity order_quantity INT NOT NULL, CHANGE order_only_missing_parts order_only_missing_parts TINYINT(1) NOT NULL, CHANGE last_modified last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, CHANGE comment comment LONGTEXT NOT NULL
     -> DROP INDEX devices_parent_id_k ON devices
     -> CREATE INDEX IDX_11074E9A727ACA70 ON devices (parent_id)
     -> ALTER TABLE devices ADD CONSTRAINT devices_parent_id_fk FOREIGN KEY (parent_id) REFERENCES devices (id)
[... 65 lines cut ...]

  ++ migrating 20190902140506 (Upgrade database from old Part-DB 0.5 Version (dbVersion 26))

     -> SET sql_mode = ''
     -> RENAME TABLE `attachement_types` TO `attachment_types`;
Migration 20190902140506 failed during Execution. Error An exception occurred while executing 'RENAME TABLE `attachement_types` TO `attachment_types`;':

[... cut ...]
  SQLSTATE[42S01]: Base table or view already exists: 1050 Table 'attachment_types' already exists

doctrine:migrations:migrate [--write-sql [WRITE-SQL]] [--dry-run] [--query-time] [--allow-no-migration] [--all-or-nothing [ALL-OR-NOTHING]] [--configuration [CONFIGURATION]] [--db-configuration [DB-CONFIGURATION]] [--db DB] [--em EM] [--shard SHARD] [-h|--help] [-q|--quiet] [-v|vv|vvv|--verbose] [-V|--version] [--ansi] [--no-ansi] [-n|--no-interaction] [-e|--env ENV] [--no-debug] [--] <command> [<version>]

coeka, 2020-06-05:
I solved this problem. One of my users has the group_id 0. After i changed it to 2 in my sqldump, this problem is fixed, but i have a new problem:

> ALTER TABLE orderdetails ADD last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, CHANGE id_supplier id_supplier INT DEFAULT NULL, CHANGE supplierpartnr supplierpartnr VARCHAR(255) NOT NULL, CHANGE obsolete obsolete TINYINT(1) NOT NULL, CHANGE supplier_product_url supplier_product_url VARCHAR(255) NOT NULL
     -> ALTER TABLE orderdetails ADD CONSTRAINT FK_489AFCDC4CE34BEC FOREIGN KEY (part_id) REFERENCES `parts` (id) ON DELETE CASCADE
     -> ALTER TABLE orderdetails ADD CONSTRAINT FK_489AFCDCCBF180EB FOREIGN KEY (id_supplier) REFERENCES `suppliers` (id)
     -> CREATE INDEX IDX_489AFCDC4CE34BEC ON orderdetails (part_id)
     -> DROP INDEX orderdetails_part_id_k ON orderdetails
     -> CREATE INDEX IDX_489AFCDCCBF180EB ON orderdetails (id_supplier)
     -> DROP INDEX orderdetails_id_supplier_k ON orderdetails
     -> DROP INDEX pricedetails_combination_uk ON pricedetails
     -> ALTER TABLE pricedetails ADD id_currency INT DEFAULT NULL, ADD datetime_added DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, CHANGE price price NUMERIC(11, 5) NOT NULL, CHANGE price_related_quantity price_related_quantity DOUBLE PRECISION NOT NULL, CHANGE min_discount_quantity min_discount_quantity DOUBLE PRECISION NOT NULL, CHANGE manual_input manual_input TINYINT(1) NOT NULL
     -> ALTER TABLE pricedetails ADD CONSTRAINT FK_C68C44594A01DDC7 FOREIGN KEY (orderdetails_id) REFERENCES `orderdetails` (id) ON DELETE CASCADE
     -> ALTER TABLE pricedetails ADD CONSTRAINT FK_C68C4459398D64AA FOREIGN KEY (id_currency) REFERENCES currencies (id)
     -> CREATE INDEX IDX_C68C4459398D64AA ON pricedetails (id_currency)
     -> CREATE INDEX IDX_C68C44594A01DDC7 ON pricedetails (orderdetails_id)
     -> DROP INDEX pricedetails_orderdetails_id_k ON pricedetails
     -> DROP INDEX name ON groups
     -> ALTER TABLE groups ADD not_selectable TINYINT(1) NOT NULL, CHANGE name name VARCHAR(255) NOT NULL, CHANGE comment comment LONGTEXT NOT NULL, CHANGE perms_labels perms_labels INT NOT NULL, CHANGE last_modified last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL
Migration 20190902140506 failed during Execution. Error An exception occurred while executing 'ALTER TABLE groups ADD not_selectable TINYINT(1) NOT NULL, CHANGE name name VARCHAR(255) NOT NULL, CHANGE comment comment LONGTEXT NOT NULL, CHANGE perms_labels perms_labels INT NOT NULL, CHANGE last_modified last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL':

SQLSTATE[42S22]: Column not found: 1054 Unknown column 'perms_labels' in 'groups'
20:09:47 ERROR     [console] Error thrown while running command "doctrine:migrations:migrate". Message: "An exception occurred while executing 'ALTER TABLE groups ADD not_selectable TINYINT(1) NOT NULL, CHANGE name name VARCHAR(255) NOT NULL, CHANGE comment comment LONGTEXT NOT NULL, CHANGE perms_labels perms_labels INT NOT NULL, CHANGE last_modified last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL':\n\nSQLSTATE[42S22]: Column not found: 1054 Unknown column 'perms_labels' in 'groups'" ["exception" => Doctrine\DBAL\Exception\InvalidFieldNameException^ { …},"command" => "doctrine:migrations:migrate","message" => """  An exception occurred while executing 'ALTER TABLE groups ADD not_selectable TINYINT(1) NOT NULL, CHANGE name name VARCHAR(255) NOT NULL, CHANGE comment comment LONGTEXT NOT NULL, CHANGE perms_labels perms_labels INT NOT NULL, CHANGE last_modified last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL':\n  \n  SQLSTATE[42S22]: Column not found: 1054 Unknown column 'perms_labels' in 'groups'  """]

In AbstractMySQLDriver.php line 60:

  An exception occurred while executing 'ALTER TABLE groups ADD not_selectable TINYINT(1) NOT NULL, CHANGE name name VARCHAR(255) NOT NULL, CHANGE comment
   comment LONGTEXT NOT NULL, CHANGE perms_labels perms_labels INT NOT NULL, CHANGE last_modified last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NUL
  L':

  SQLSTATE[42S22]: Column not found: 1054 Unknown column 'perms_labels' in 'groups'

In PDOConnection.php line 83:

  SQLSTATE[42S22]: Column not found: 1054 Unknown column 'perms_labels' in 'groups'

In PDOConnection.php line 78:

  SQLSTATE[42S22]: Column not found: 1054 Unknown column 'perms_labels' in 'groups'

coeka, 2020-06-05:
I changed line 178 in Version20190902140506.php, so that the column perms_labels is added:

$this->addSql('ALTER TABLE groups ADD not_selectable TINYINT(1) NOT NULL, CHANGE name name VARCHAR(255) NOT NULL, CHANGE comment comment LONGTEXT NOT NULL, add perms_labels INT NOT NULL, CHANGE last_modified last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL');

With this change the migration was successfull. Do i need the column perms_labels in the table groups?

jbtronics, 2020-06-07:
From which old Part-DB  version has you upgraded?

Could you send me a dump of the database (before upgrade)? I guess there is something not the way the migrations expect it.

Also when an exception happens during the migrations, you should import your backup, to start over freshly. It is really difficult to debug migrations that was only partly done (and MySQL does not support transactions for schema changes, so there is no way to revert the changes automatically).

jbtronics, 2020-07-06:
I have sent you an email.

I guess this is an issue specific to your database, therefore I am closing this issue.

closed 2020-07-06 by jbtronics

coeka, 2021-01-11:
I think the problem is, that my first Part-DB-Version was 0.59 or 0.6.

When the old version creates the database in Part-DB/updates/db_update_steps.php
only Case 0 is executed because of the break; in line 532

Because of this, there are several differences between my database and the database, which you expect with version 26.