Part-DB/Part-DB-server #74: Drop index on groups failes when upgrading from part-db-0.6.0 to 1.0

NightDragon1, 2020-06-30:
**Describe the bug**
When running ` php bin/console doctrine:migrations:migrate` to upgrade my part-db-0.6.0 to 1.0 I run into an issue when it tried to drop the index "name" on table "groups".

Additional Info:
I use the docker-instance of partdb-symfony as given in the hint. The APP_ENV was set to prod (docker env as also the .env.local file) and .env.local was updated to point to a copy of my old databse.

**To Reproduce**
1. Update .env.local to point to the partdb-0.6.0 database.
2.  run ` php bin/console doctrine:migrations:migrate`
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Output Migration script**
```
php bin/console doctrine:migrations:migrate
                                                              
                    Application Migrations                    
                                                              

WARNING! You are about to execute a database migration that could result in schema changes and data loss. Are you sure you wish to continue? (y/n)y
Migrating up to 20200502161750 from 0

  ++ migrating 1 (Creates an inital empty database)

  SS skipped (Reason: Old Part-DB Database detected! Continue with upgrade...)

  ++ migrating 20190902140506 (Upgrade database from old Part-DB 0.5 Version (dbVersion 26))

     -> SET sql_mode = ''
     -> RENAME TABLE `attachement_types` TO `attachment_types`;
     -> RENAME TABLE `attachements` TO `attachments`;
     -> CREATE TABLE currencies (id INT AUTO_INCREMENT NOT NULL, parent_id INT DEFAULT NULL, iso_code VARCHAR(255) NOT NULL, exchange_rate NUMERIC(11, 5) DEFAULT NULL, comment LONGTEXT NOT NULL, not_selectable TINYINT(1) NOT NULL, name VARCHAR(255) NOT NULL, last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, datetime_added DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, INDEX IDX_37C44693727ACA70 (parent_id), PRIMARY KEY(id)) DEFAULT CHARACTER SET utf8mb4 COLLATE utf8mb4_unicode_ci ENGINE = InnoDB
     -> CREATE TABLE `measurement_units` (id INT AUTO_INCREMENT NOT NULL, parent_id INT DEFAULT NULL, unit VARCHAR(255) DEFAULT NULL, is_integer TINYINT(1) NOT NULL, use_si_prefix TINYINT(1) NOT NULL, comment LONGTEXT NOT NULL, not_selectable TINYINT(1) NOT NULL, name VARCHAR(255) NOT NULL, last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, datetime_added DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, INDEX IDX_F5AF83CF727ACA70 (parent_id), PRIMARY KEY(id)) DEFAULT CHARACTER SET utf8mb4 COLLATE utf8mb4_unicode_ci ENGINE = InnoDB
     -> CREATE TABLE part_lots (id INT AUTO_INCREMENT NOT NULL, id_store_location INT DEFAULT NULL, id_part INT NOT NULL, description LONGTEXT NOT NULL, comment LONGTEXT NOT NULL, expiration_date DATETIME DEFAULT NULL, instock_unknown TINYINT(1) NOT NULL, amount DOUBLE PRECISION NOT NULL, needs_refill TINYINT(1) NOT NULL, last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, datetime_added DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, INDEX IDX_EBC8F9435D8F4B37 (id_store_location), INDEX IDX_EBC8F943C22F6CC4 (id_part), PRIMARY KEY(id)) DEFAULT CHARACTER SET utf8mb4 COLLATE utf8mb4_unicode_ci ENGINE = InnoDB
     -> INSERT IGNORE INTO part_lots (id_part, id_store_location, amount, instock_unknown, last_modified, datetime_added) SELECT parts.id, parts.id_storelocation,  parts.instock, 0, NOW(), NOW() FROM parts WHERE parts.instock >= 0
     -> INSERT IGNORE INTO part_lots (id_part, id_store_location, amount, instock_unknown, last_modified, datetime_added) SELECT parts.id, parts.id_storelocation, 0, 1, NOW(), NOW() FROM parts WHERE parts.instock = -2
     -> ALTER TABLE currencies ADD CONSTRAINT FK_37C44693727ACA70 FOREIGN KEY (parent_id) REFERENCES currencies (id)
     -> ALTER TABLE `measurement_units` ADD CONSTRAINT FK_F5AF83CF727ACA70 FOREIGN KEY (parent_id) REFERENCES `measurement_units` (id)
     -> ALTER TABLE part_lots ADD CONSTRAINT FK_EBC8F9435D8F4B37 FOREIGN KEY (id_store_location) REFERENCES `storelocations` (id)
     -> ALTER TABLE part_lots ADD CONSTRAINT FK_EBC8F943C22F6CC4 FOREIGN KEY (id_part) REFERENCES `parts` (id) ON DELETE CASCADE
     -> ALTER TABLE parts DROP INDEX parts_order_orderdetails_id_k, ADD UNIQUE INDEX UNIQ_6940A7FE81081E9B (order_orderdetails_id)
     -> ALTER TABLE parts DROP FOREIGN KEY parts_id_storelocation_fk
     -> DROP INDEX favorite ON parts
     -> DROP INDEX parts_id_storelocation_k ON parts
     -> ALTER TABLE parts DROP FOREIGN KEY parts_id_footprint_fk
     -> ALTER TABLE parts DROP FOREIGN KEY parts_id_manufacturer_fk
     -> ALTER TABLE parts CHANGE mininstock minamount DOUBLE PRECISION NOT NULL, ADD id_part_unit INT DEFAULT NULL, ADD manufacturer_product_number VARCHAR(255) NOT NULL, ADD manufacturing_status VARCHAR(255) DEFAULT NULL, ADD needs_review TINYINT(1) NOT NULL, ADD tags LONGTEXT NOT NULL, ADD mass DOUBLE PRECISION DEFAULT NULL, DROP instock, CHANGE id_category id_category INT NOT NULL, CHANGE name name VARCHAR(255) NOT NULL, CHANGE order_quantity order_quantity INT NOT NULL, CHANGE manual_order manual_order TINYINT(1) NOT NULL, CHANGE manufacturer_product_url manufacturer_product_url VARCHAR(255) NOT NULL, CHANGE last_modified last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, CHANGE favorite favorite TINYINT(1) NOT NULL, DROP id_storelocation
     -> ALTER TABLE parts ADD CONSTRAINT FK_6940A7FE5697F554 FOREIGN KEY (id_category) REFERENCES `categories` (id)
     -> ALTER TABLE parts ADD CONSTRAINT FK_6940A7FEEBBCC786 FOREIGN KEY (id_master_picture_attachement) REFERENCES `attachments` (id)
     -> ALTER TABLE parts ADD CONSTRAINT FK_6940A7FE2626CEF9 FOREIGN KEY (id_part_unit) REFERENCES `measurement_units` (id)
     -> CREATE INDEX IDX_6940A7FEEBBCC786 ON parts (id_master_picture_attachement)
     -> CREATE INDEX IDX_6940A7FE2626CEF9 ON parts (id_part_unit)
     -> CREATE INDEX IDX_6940A7FE5697F554 ON parts (id_category)
     -> DROP INDEX parts_id_category_k ON parts
     -> DROP INDEX parts_id_footprint_k ON parts
     -> CREATE INDEX IDX_6940A7FE7E371A10 ON parts (id_footprint)
     -> DROP INDEX parts_id_manufacturer_k ON parts
     -> CREATE INDEX IDX_6940A7FE1ECB93AE ON parts (id_manufacturer)
     -> ALTER TABLE parts ADD CONSTRAINT parts_id_footprint_fk FOREIGN KEY (id_footprint) REFERENCES footprints (id)
     -> ALTER TABLE parts ADD CONSTRAINT parts_id_manufacturer_fk FOREIGN KEY (id_manufacturer) REFERENCES manufacturers (id)
     -> ALTER TABLE attachment_types DROP FOREIGN KEY attachement_types_parent_id_fk
     -> ALTER TABLE attachment_types ADD filetype_filter LONGTEXT NOT NULL, ADD not_selectable TINYINT(1) NOT NULL, CHANGE name name VARCHAR(255) NOT NULL, CHANGE comment comment LONGTEXT NOT NULL, CHANGE last_modified last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL
     -> DROP INDEX attachement_types_parent_id_k ON attachment_types
     -> CREATE INDEX IDX_EFAED719727ACA70 ON attachment_types (parent_id)
     -> ALTER TABLE attachment_types ADD CONSTRAINT attachement_types_parent_id_fk FOREIGN KEY (parent_id) REFERENCES attachment_types (id)
     -> ALTER TABLE categories DROP FOREIGN KEY categories_parent_id_fk
     -> ALTER TABLE categories ADD not_selectable TINYINT(1) NOT NULL, CHANGE name name VARCHAR(255) NOT NULL, CHANGE disable_footprints disable_footprints TINYINT(1) NOT NULL, CHANGE disable_manufacturers disable_manufacturers TINYINT(1) NOT NULL, CHANGE disable_autodatasheets disable_autodatasheets TINYINT(1) NOT NULL, CHANGE disable_properties disable_properties TINYINT(1) NOT NULL, CHANGE comment comment LONGTEXT NOT NULL, CHANGE last_modified last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL
     -> DROP INDEX categories_parent_id_k ON categories
[... 83 lines cut ...]
  SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use nea  
  r 'groups' at line 1                                                                                                                                                                          
                                                                                                                                                                                                

In PDOConnection.php line 78:
[... cut ...]
**Smartphone (please complete the following information):**
N/A

**Additional context**
I already tried to drop the index manually and get the following message back `Error 1067: Invalid default value for 'last_modified' SQL Statement: ALTER TABLE `partdb`.`groups`  CHANGE COLUMN `last_modified` `last_modified` TIMESTAMP NOT NULL DEFAULT '00-00-0000 00:00:00'   
`
I also tried to allow 'last_modified' to be NULL and set the default value to NULL without success. The drop is still not allowed. I'm a bit clueless...

coeka, 2020-07-01:
I think this could be related to MariaDB version.

Requirements:
A MySQL (at least 5.6.5) /MariaDB (at least 10.0.1) database server if you do not want to use SQLite.

and 

https://github.com/Part-DB/Part-DB-symfony/issues/8

NightDragon1, 2020-07-01:
Sorry guys. I gave you wrong info. It's mysql not mariadb.

NightDragon1, 2020-07-01:
Using mariadb fixed the issue.

jbtronics, 2020-07-04:
Part-DB should work on MySQL too, so i reopen the issue...

closed 2023-07-08 by jbtronics