Part-DB/Part-DB-server #75: Error when executing doctrine:migrations:migrate

tbnobody, 2020-07-05:
**Describe the bug**
When executed `doctrine:migrations:migrate` I got the following error message:
```
03:38:13 ERROR     [console] Error thrown while running command "doctrine:migrations:migrate". Message: "The metadata storage is not up to date, please run the sync-metadata-storage command to fix rine\Migrations\Exception\MetadataStorageError^ { …},"command" => "doctrine:migrations:migrate","message" => "The metadata storage is not up to date, please run the sync-metadata-storage command to

In MetadataStorageError.php line 13:

  The metadata storage is not up to date, please run the sync-metadata-storage command to fix this issue.

doctrine:migrations:migrate [--write-sql [WRITE-SQL]] [--dry-run] [--query-time] [--allow-no-migration] [--all-or-nothing [ALL-OR-NOTHING]] [--configuration CONFIGURATION] [--db-configuration DB-COiet] [-v|vv|vvv|--verbose] [-V|--version] [--ansi] [--no-ansi] [-n|--no-interaction] [-e|--env ENV] [--no-debug] [--] <command> [<version>]
```

At first when I executed `doctrine:migrations:sync-metadata-storage` I got the same exception. I fixed this by setting the right serverVersion in the .env.local file:
```
sudo -u www-data bin/console doctrine:migrations:sync-metadata-storage -vvv

 [OK] Metadata storage synchronized
```

When then executing doctrine:migrations:migrate I got the following error message:
```
sudo -u www-data php bin/console doctrine:migrations:migrate                                                                                                           
 WARNING! You are about to execute a database migration that could result in schema changes and data loss. Are you sure you wish to continue? (yes/no) [yes]:
 >

10:59:57 ERROR     [app] Migration DoctrineMigrations\Version1 failed during Execution. Error: "An exception occurred while executing 'CREATE TABLE `attachment_types` (id INT AUTO_INCREMENT NOT NULL, parent_id INT DEFAULT NULL, filetype_filter LONGTEXT NOT NULL, comment LONGTEXT NOT NULL, not_selectable TINYINT(1) NOT NULL, name VARCHAR(255) NOT NULL, last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, datetime_added DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, INDEX IDX_EFAED719727ACA70 (parent_id), PRIMARY KEY(id)) DEFAULT CHARACTER SET utf8mb4 COLLATE utf8mb4_unicode_ci ENGINE = InnoDB':\n\nSQLSTATE[42S01]: Base table or view already exists: 1050 Table 'attachment_types' already exists" ["version" => "DoctrineMigrations\Version1","error" => """  An exception occurred while executing 'CREATE TABLE `attachment_types` (id INT AUTO_INCREMENT NOT NULL, parent_id INT DEFAULT NULL, filetype_filter LONGTEXT NOT NULL, comment LONGTEXT NOT NULL, not_selectable TINYINT(1) NOT NULL, name VARCHAR(255) NOT NULL, last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, datetime_added DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, INDEX IDX_EFAED719727ACA70 (parent_id), PRIMARY KEY(id)) DEFAULT CHARACTER SET utf8mb4 COLLATE utf8mb4_unicode_ci ENGINE = InnoDB':\n  \n  SQLSTATE[42S01]: Base table or view already exists: 1050 Table 'attachment_types' already exists  ""","state" => "Execution"]
10:59:57 ERROR     [console] Error thrown while running command "doctrine:migrations:migrate". Message: "An exception occurred while executing 'CREATE TABLE `attachment_types` (id INT AUTO_INCREMENT NOT NULL, parent_id INT DEFAULT NULL, filetype_filter LONGTEXT NOT NULL, comment LONGTEXT NOT NULL, not_selectable TINYINT(1) NOT NULL, name VARCHAR(255) NOT NULL, last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, datetime_added DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, INDEX IDX_EFAED719727ACA70 (parent_id), PRIMARY KEY(id)) DEFAULT CHARACTER SET utf8mb4 COLLATE utf8mb4_unicode_ci ENGINE = InnoDB':\n\nSQLSTATE[42S01]: Base table or view already exists: 1050 Table 'attachment_types' already exists" ["exception" => Doctrine\DBAL\Exception\TableExistsException^ { …},"command" => "doctrine:migrations:migrate","message" => """  An exception occurred while executing 'CREATE TABLE `attachment_types` (id INT AUTO_INCREMENT NOT NULL, parent_id INT DEFAULT NULL, filetype_filter LONGTEXT NOT NULL, comment LONGTEXT NOT NULL, not_selectable TINYINT(1) NOT NULL, name VARCHAR(255) NOT NULL, last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, datetime_added DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, INDEX IDX_EFAED719727ACA70 (parent_id), PRIMARY KEY(id)) DEFAULT CHARACTER SET utf8mb4 COLLATE utf8mb4_unicode_ci ENGINE = InnoDB':\n  \n  SQLSTATE[42S01]: Base table or view already exists: 1050 Table 'attachment_types' already exists  """]

In AbstractMySQLDriver.php line 38:

  An exception occurred while executing 'CREATE TABLE `attachment_types` (id INT AUTO_INCREMENT NOT NULL, parent_id INT DEFAULT NULL, filetype_filter LONGTEXT NOT NULL, comment LONGTEXT NOT NULL
  , not_selectable TINYINT(1) NOT NULL, name VARCHAR(255) NOT NULL, last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, datetime_added DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, INDEX I
  DX_EFAED719727ACA70 (parent_id), PRIMARY KEY(id)) DEFAULT CHARACTER SET utf8mb4 COLLATE utf8mb4_unicode_ci ENGINE = InnoDB':

  SQLSTATE[42S01]: Base table or view already exists: 1050 Table 'attachment_types' already exists

In PDOConnection.php line 43:

  SQLSTATE[42S01]: Base table or view already exists: 1050 Table 'attachment_types' already exists

In PDOConnection.php line 41:

  SQLSTATE[42S01]: Base table or view already exists: 1050 Table 'attachment_types' already exists

doctrine:migrations:migrate [--write-sql [WRITE-SQL]] [--dry-run] [--query-time] [--allow-no-migration] [--all-or-nothing [ALL-OR-NOTHING]] [--configuration CONFIGURATION] [--db-configuration DB-CONFIGURATION] [-h|--help] [-q|--quiet] [-v|vv|vvv|--verbose] [-V|--version] [--ansi] [--no-ansi] [-n|--no-interaction] [-e|--env ENV] [--no-debug] [--] <command> [<version>]
```

I found out that I have two different tables in the database: `doctrine_migration_versions` and `migration_versions`. The first one is empty, the second one contains the already executed migrations:
```
+----------------+---------------------+
| version        | executed_at         |
+----------------+---------------------+
| 1              | 2020-04-06 11:35:50 |
| 20190902140506 | 2020-04-06 11:35:50 |
| 20190913141126 | 2020-04-06 11:35:50 |
| 20190924113252 | 2020-04-06 11:35:52 |
| 20191214153125 | 2020-04-06 11:35:52 |
| 20200126191823 | 2020-04-06 11:35:52 |
| 20200311204104 | 2020-04-06 11:35:52 |
| 20200409130946 | 2020-04-13 19:31:11 |
| 20200502161750 | 2020-05-10 12:43:42 |
+----------------+---------------------+
```

**Server Side**
- Part-DB Version: a02526cd787d12d3e256fe35a646aa325273cca0
- PHP Version: `PHP 7.3.19-1+0~20200612.60+debian10~1.gbp6c8fe1 (cli) (built: Jun 12 2020 07:48:33) ( NTS )`
- Database Server `10.3.22-MariaDB-0+deb10u1 for debian-linux-gnu on x86_64 (Debian 10)`

jbtronics, 2020-07-06:
Should be fixed.

rckTom, 2020-07-08:
This does not fix the issue for me. Trying to update to latest master (1a39d80) i get the following errors:
```
 WARNING! You are about to execute a database migration that could result in schema changes and data loss. Are you sure you wish to continue? (yes/no) [yes]:
 > y

                                                                                                                      
 [WARNING] You have 9 previously executed migrations in the database that are not registered migrations.              
                                                                                                                      

 >> 2020-05-30 17:32:46 (1)
 >> 2020-05-30 17:32:46 (20190902140506)
 >> 2020-05-30 17:32:47 (20190913141126)
 >> 2020-05-30 17:32:47 (20190924113252)
 >> 2020-05-30 17:32:48 (20191214153125)
 >> 2020-05-30 17:32:48 (20200126191823)
 >> 2020-05-30 17:32:48 (20200311204104)
 >> 2020-05-30 17:32:48 (20200409130946)
 >> 2020-05-30 17:32:48 (20200502161750)

 Are you sure you wish to continue? (yes/no) [yes]:
 > y

16:27:17 ERROR     [app] Migration DoctrineMigrations\Version1 failed during Execution. Error: "An exception occurred while executing 'CREATE TABLE `attachment_types` (id INT AUTO_INCREMENT NOT NULL, parent_id INT DEFAULT NULL, filetype_filter LONGTEXT NOT NULL, comment LONGTEXT NOT NULL, not_selectable TINYINT(1) NOT NULL, name VARCHAR(255) NOT NULL, last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, datetime_added DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, INDEX IDX_EFAED719727ACA70 (parent_id), PRIMARY KEY(id)) DEFAULT CHARACTER SET utf8mb4 COLLATE utf8mb4_unicode_ci ENGINE = InnoDB':\n\nSQLSTATE[42S01]: Base table or view already exists: 1050 Table 'attachment_types' already exists" ["version" => "DoctrineMigrations\Version1","error" => """  An exception occurred while executing 'CREATE TABLE `attachment_types` (id INT AUTO_INCREMENT NOT NULL, parent_id INT DEFAULT NULL, filetype_filter LONGTEXT NOT NULL, comment LONGTEXT NOT NULL, not_selectable TINYINT(1) NOT NULL, name VARCHAR(255) NOT NULL, last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, datetime_added DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, INDEX IDX_EFAED719727ACA70 (parent_id), PRIMARY KEY(id)) DEFAULT CHARACTER SET utf8mb4 COLLATE utf8mb4_unicode_ci ENGINE = InnoDB':\n  \n  SQLSTATE[42S01]: Base table or view already exists: 1050 Table 'attachment_types' already exists  ""","state" => "Execution"]
16:27:17 ERROR     [console] Error thrown while running command "doctrine:migrations:migrate". Message: "An exception occurred while executing 'CREATE TABLE `attachment_types` (id INT AUTO_INCREMENT NOT NULL, parent_id INT DEFAULT NULL, filetype_filter LONGTEXT NOT NULL, comment LONGTEXT NOT NULL, not_selectable TINYINT(1) NOT NULL, name VARCHAR(255) NOT NULL, last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, datetime_added DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, INDEX IDX_EFAED719727ACA70 (parent_id), PRIMARY KEY(id)) DEFAULT CHARACTER SET utf8mb4 COLLATE utf8mb4_unicode_ci ENGINE = InnoDB':\n\nSQLSTATE[42S01]: Base table or view already exists: 1050 Table 'attachment_types' already exists" ["exception" => Doctrine\DBAL\Exception\TableExistsException^ { …},"command" => "doctrine:migrations:migrate","message" => """  An exception occurred while executing 'CREATE TABLE `attachment_types` (id INT AUTO_INCREMENT NOT NULL, parent_id INT DEFAULT NULL, filetype_filter LONGTEXT NOT NULL, comment LONGTEXT NOT NULL, not_selectable TINYINT(1) NOT NULL, name VARCHAR(255) NOT NULL, last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, datetime_added DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, INDEX IDX_EFAED719727ACA70 (parent_id), PRIMARY KEY(id)) DEFAULT CHARACTER SET utf8mb4 COLLATE utf8mb4_unicode_ci ENGINE = InnoDB':\n  \n  SQLSTATE[42S01]: Base table or view already exists: 1050 Table 'attachment_types' already exists  """]

In AbstractMySQLDriver.php line 38:
                                                                                                                     
  An exception occurred while executing 'CREATE TABLE `attachment_types` (id INT AUTO_INCREMENT NOT NULL, parent_id  
   INT DEFAULT NULL, filetype_filter LONGTEXT NOT NULL, comment LONGTEXT NOT NULL, not_selectable TINYINT(1) NOT NU  
  LL, name VARCHAR(255) NOT NULL, last_modified DATETIME DEFAULT CURRENT_TIMESTAMP NOT NULL, datetime_added DATETIM  
  E DEFAULT CURRENT_TIMESTAMP NOT NULL, INDEX IDX_EFAED719727ACA70 (parent_id), PRIMARY KEY(id)) DEFAULT CHARACTER   
  SET utf8mb4 COLLATE utf8mb4_unicode_ci ENGINE = InnoDB':                                                           
                                                                                                                     
  SQLSTATE[42S01]: Base table or view already exists: 1050 Table 'attachment_types' already exists                   
                                                                                                                     

In PDOConnection.php line 43:
                                                                                                    
  SQLSTATE[42S01]: Base table or view already exists: 1050 Table 'attachment_types' already exists  
                                                                                                    

In PDOConnection.php line 41:
                                                                                                    
[... cut ...]
| 20200126191823 | 2020-05-30 17:32:48 |           NULL |
| 20200311204104 | 2020-05-30 17:32:48 |           NULL |
| 20200409130946 | 2020-05-30 17:32:48 |           NULL |
| 20200502161750 | 2020-05-30 17:32:48 |           NULL |
+----------------+---------------------+----------------+
```

Setup:
* PartDB-version: 6b8d06466687f911bca7add92115c6c4475f886e
* PHP-version: PHP 7.3.19-1~deb10u1 (cli) (built: Jul  5 2020 06:46:45) ( NTS )
* Database: mariadb  Ver 15.1 Distrib 10.3.22-MariaDB, for debian-linux-gnu (x86_64) using readline 5.2

tbnobody, 2021-05-08:
Not sure whether this issue occured for other users but  we fixed it some time ago by manually adjusting the version column in migration_versions table. (To be honest, I don't know anymore what exactly I did). From our side (@rckTom and me) this issue can be closed.

Xernito, 2021-08-21:
The Error is: Base table or view already exists: 1050 Table 'attachment_types' **already exists** 

So drop the Database and run the command again.

tbnobody, 2021-09-27:
And all the data which is maybe already inserted would be gone....