Part-DB/Part-DB-server #82: php bin/console app:convert-bbcode

OrionAC, 2020-09-28:
**Describe the bug**
when migrating the database

**Expected behavior**

php bin/console app:convert-bbcode

Convert entities of class App\Entity\Parts\Part
-----------------------------------------------

 ! [NOTE] Search for entities of type App\Entity\Parts\Part that need !        conversion

10:55:28 ERROR     [console] Error thrown while running command "app:convert-bbc
ode". Message: "An exception occurred while executing 'SELECT p0_.datetime_added
 AS datetime_added_0, p0_.name AS name_1, p0_.last_modified AS last_modified_2,
p0_.id AS id_3, p0_.needs_review AS needs_review_4, p0_.tags AS tags_5, p0_.mass
 AS mass_6, p0_.description AS description_7, p0_.comment AS comment_8, p0_.visi
ble AS visible_9, p0_.favorite AS favorite_10, p0_.minamount AS minamount_11, p0
_.manufacturer_product_url AS manufacturer_product_url_12, p0_.manufacturer_prod
uct_number AS manufacturer_product_number_13, p0_.manufacturing_status AS manufa
cturing_status_14, p0_.order_quantity AS order_quantity_15, p0_.manual_order AS
manual_order_16, p0_.id_preview_attachement AS id_preview_attachement_17, p0_.id
_category AS id_category_18, p0_.id_footprint AS id_footprint_19, p0_.id_part_un
it AS id_part_unit_20, p0_.id_manufacturer AS id_manufacturer_21, p0_.order_orde
rdetails_id AS order_orderdetails_id_22 FROM `parts` p0_ WHERE p0_.description L
IKE ? OR p0_.comment LIKE ?' with params ["%[%]%[\/%]%", "%[%]%[\/%]%"]:\n\nSQLS
TATE[42S22]: Column not found: 1054 Unknown column 'p0_.needs_review' in 'field
list'" ["exception" => Doctrine\DBAL\Exception\InvalidFieldNameException { ÔÇª},
"command" => "app:convert-bbcode","message" => """  An exception occurred while
executing 'SELECT p0_.datetime_added AS datetime_added_0, p0_.name AS name_1, p0
_.last_modified AS last_modified_2, p0_.id AS id_3, p0_.needs_review AS needs_re
view_4, p0_.tags AS tags_5, p0_.mass AS mass_6, p0_.description AS description_7
, p0_.comment AS comment_8, p0_.visible AS visible_9, p0_.favorite AS favorite_1
0, p0_.minamount AS minamount_11, p0_.manufacturer_product_url AS manufacturer_p
roduct_url_12, p0_.manufacturer_product_number AS manufacturer_product_number_13
, p0_.manufacturing_status AS manufacturing_status_14, p0_.order_quantity AS ord
er_quantity_15, p0_.manual_order AS manual_order_16, p0_.id_preview_attachement
AS id_preview_attachement_17, p0_.id_category AS id_category_18, p0_.id_footprin
t AS id_footprint_19, p0_.id_part_unit AS id_part_unit_20, p0_.id_manufacturer A
S id_manufacturer_21, p0_.order_orderdetails_id AS order_orderdetails_id_22 FROM
 `parts` p0_ WHERE p0_.description LIKE ? OR p0_.comment LIKE ?' with params ["%
[%]%[\/%]%", "%[%]%[\/%]%"]:\n  \n  SQLSTATE[42S22]: Column not found: 1054 Unkn
own column 'p0_.needs_review' in 'field list'  """]

In AbstractMySQLDriver.php line 64:

  An exception occurred while executing 'SELECT p0_.datetime_added AS datetim
  e_added_0, p0_.name AS name_1, p0_.last_modified AS last_modified_2, p0_.id
   AS id_3, p0_.needs_review AS needs_review_4, p0_.tags AS tags_5, p0_.mass
  AS mass_6, p0_.description AS description_7, p0_.comment AS comment_8, p0_.
  visible AS visible_9, p0_.favorite AS favorite_10, p0_.minamount AS minamou
  nt_11, p0_.manufacturer_product_url AS manufacturer_product_url_12, p0_.man
  ufacturer_product_number AS manufacturer_product_number_13, p0_.manufacturi
  ng_status AS manufacturing_status_14, p0_.order_quantity AS order_quantity_
  15, p0_.manual_order AS manual_order_16, p0_.id_preview_attachement AS id_p
  review_attachement_17, p0_.id_category AS id_category_18, p0_.id_footprint
  AS id_footprint_19, p0_.id_part_unit AS id_part_unit_20, p0_.id_manufacture
  r AS id_manufacturer_21, p0_.order_orderdetails_id AS order_orderdetails_id
  _22 FROM `parts` p0_ WHERE p0_.description LIKE ? OR p0_.comment LIKE ?' wi
  th params ["%[%]%[\/%]%", "%[%]%[\/%]%"]:

  SQLSTATE[42S22]: Column not found: 1054 Unknown column 'p0_.needs_review' i
  n 'field list'

In PDOStatement.php line 131:

  SQLSTATE[42S22]: Column not found: 1054 Unknown column 'p0_.needs_review' i
  n 'field list'

In PDOStatement.php line 129:

  SQLSTATE[42S22]: Column not found: 1054 Unknown column 'p0_.needs_review' i
  n 'field list'

app:convert-bbcode [--dry-run] [-h|--help] [-q|--quiet] [-v|vv|vvv|--verbose] [-
V|--version] [--ansi] [--no-ansi] [-n|--no-interaction] [-e|--env ENV] [--no-deb
ug] [--] <command>

**Server Side**
- Part-DB Version: [Part-DB 1.0]
- PHP Version: [PHP 7.4.10]
- Database Server [MySQL 7.4.10]

**Desktop (please complete the following information):**
 - OS: [Win 7]
 - Browser [Crom]
 - Version [?]

jbtronics, 2020-09-28:
Has you run `php bin/console doctrine:migrations:migrate` before? and was it successful?

This looks like it has something to do with a wrong (or a not up-to-date) database.

UtechtDustin, 2021-11-23:
@jbtronics 
We also have the issue, currently we are using Part-DB Version 0.5.9
I duplicated the database and tried to upgrade, but i get the following error if i run `php bin/console doctrine:migrations:migrate`

```

 WARNING! You are about to execute a migration in database "partdb-live-up" that could result in schema changes and data loss. Are you sure you wish to continue? (yes/no) [yes]:
 > yes

[notice] Migrating up to DoctrineMigrations\Version20200502161750
[warning] Old Part-DB Database detected! Continue with upgrade...
[warning] Migration DoctrineMigrations\Version1 was executed but did not result in any SQL statements.
[warning] [!!!] Permissions were updated! Please check if they fit your expectations!
[error] Migration DoctrineMigrations\Version20190902140506 failed during Execution. Error: "An exception occurred while executing 'ALTER TABLE parts DROP FOREIGN KEY parts_id_footprint_fk':

SQLSTATE[42000]: Syntax error or access violation: 1091 Can't DROP 'parts_id_footprint_fk'; check that column/key exists"
15:00:59 CRITICAL  [console] Error thrown while running command "doctrine:migrations:migrate". Message: "An exception occurred while executing 'ALTER TABLE parts DROP FOREIGN KEY parts_id_footprint_fk':\n\nSQLSTATE[42000]: Syntax error or access violation: 1091 Can't DROP 'parts_id_footprint_fk'; check that column/key exists" ["exception" => Doctrine\DBAL\Exception\DriverException^ { …},"command" => "doctrine:migrations:migrate","message" => """  An exception occurred while executing 'ALTER TABLE parts DROP FOREIGN KEY parts_id_footprint_fk':\n  \n  SQLSTATE[42000]: Syntax error or access violation: 1091 Can't DROP 'parts_id_footprint_fk'; check that column/key exists  """]

In AbstractMySQLDriver.php line 128:

  An exception occurred while executing 'ALTER TABLE parts DROP FOREIGN KEY parts_id_footprint_fk':

  SQLSTATE[42000]: Syntax error or access violation: 1091 Can't DROP 'parts_id_footprint_fk'; check that column/key exists

In Exception.php line 18:

  SQLSTATE[42000]: Syntax error or access violation: 1091 Can't DROP 'parts_id_footprint_fk'; check that column/key exists

In PDOConnection.php line 137:

  SQLSTATE[42000]: Syntax error or access violation: 1091 Can't DROP 'parts_id_footprint_fk'; check that column/key exists

doctrine:migrations:migrate [--write-sql [WRITE-SQL]] [--dry-run] [--query-time] [--allow-no-migration] [--all-or-nothing [ALL-OR-NOTHING]] [--configuration CONFIGURATION] [--em EM] [--conn CONN] [--] [<version>]
```
I tried to remove the `DROP FOREIGN KEY`-Part, but after i removed 5 `DROP FOREIGN KEY` i stopped.

If i skip  `php bin/console doctrine:migrations:migrate` and run `php bin/console app:convert-bbcode
` directly i get the same issue as @OrionAC.

jbtronics, 2023-01-28:
You have to do the database migrations before attempting to do BBCode conversion.

closed 2023-01-28 by jbtronics