Part-DB/Part-DB-server #86: backup/restore documentation

927589452, 2020-11-23:
**Is your feature request related to a problem? Please describe.**
The old part-db had documentation on what was needed to backup an what was needed to restore

**Describe the solution you'd like**
List of files / directories to backup 

**Describe alternatives you've considered**
A php script to create an backup archive like `php bin/console backup:{create, restore}`

jbtronics, 2023-01-28:
(Basic) Backup and Restore instructions can now be found here: https://github.com/Part-DB/Part-DB-symfony/wiki/Backup-&-Restore-Data.

Also there now is a command `php bin/console partdb:backup` which creates a ZIP file with all back-uped data (e.g. `php bin/console/partdb:backup --full /root/backup.zip`

closed 2023-01-28 by jbtronics

virtadpt, 2023-01-29:
Woot!  Thank you!