Part-DB/Part-DB-server #89: Server gets stuck on 500 Internal Server Error if unsupported image attached to part

stenstad, 2021-01-05:
If you upload any image not supported by gd.so, the server will throw an execption in /var/www/html/vendor/imagine/imagine/src/Gd/Imagine.php at line 95, since createImageFromString($data) above it only creates a valid resource if GD supports the image format.

The GD bundled with the docker image (jbtronics/part-db1:latest) does not support a lot of formats. PNG works, JPEG or WEBP does not.

It should probably fail more gracefully, as you need to delete the file on disk to get PartDB running again.

**To Reproduce**
Steps to reproduce the behavior:
1. Add an attachment to any part.
2. Upload a jpeg.
3. Wonder why PHP is not able to access the file on disk.

Chris-42, 2021-08-29:
seema again to be a problem

jbtronics, 2022-07-20:
Part-DB should dont throw an exception which makes it impossible to continue now. Instead, the original file is served, if no thumbnail can be generated.

closed 2022-07-23 by jbtronics