Part-DB/Part-DB-server #91: 1D Barcode subtitle is printed behind Barcode

Marioheld, 2021-01-07:
When trying to print 1D Barcodes (Code 128, Code 39 or Code 93) on labels (Storage location, Part lot, or Part) the subtilte of the barcode is printed behind the barcode itself. 
Are there any ways in the label generator to modify the barcode layout? 

**Screenshot**
[image]

**Tested Versions**
- Latest checkout from master branch on Debian 10
- Heroku Instance EN and DE

Mellerin, 2021-03-14:
Hello,
I am currently trying out Part-DB with the same Heroku(en) instance, and I could see this while testing this feature (with Firefox latest version, didn't try another browser).The Part-DB-symfony/templates/LabelSystem/labels/label_style.css.twig file contains the CSS for rendering this label.
When creating a label, you can change the style in the "Advanced" tab, I copy-pasted the style defined in the said file and modified it like this:

```
.C39-container {
    display: inline-block;
    align-content: center;
    text-align: left;
    position: fixed;
    bottom: 50px;
}

.C39-text {
    display: block;
    margin-top: 6px;
    font-size: 6pt;
}
```
I don't know much about CSS or styles, but by playing with the values I found a rendering that works with Code 39/93/128.
[image]

@ dev and all contributor : thanks for this promising app :)

jbtronics, 2023-01-29:
This was fixed some time ago.

closed 2023-01-29 by jbtronics