Part-DB/Part-DB-server #99: Mobile optimized page for quick and easy stock management

reichemn, 2021-04-24:
Hi,
i would love it if the app would provide a page which can be used at a mobile phone in combination with printed barcode labels to quickly manage the stock and storage locations of parts. This page could you allow to scan the label of a part and provide actions to:

- Reduce/Increase the stock in some storage location (choose location from list or scan label)
- Scan the label from a storage location which was never used for this part to add it to the part and set the stock value
- Move stock between two storage locations (choose locations from list or scan label)

reichemn, 2023-03-26:
If the server would offer an API to manage the stocks and storage related things, it would be very little effort for someone to build a mobile app for this use case.
It think more documented APIs would make this project more powerful in general, because users could write them self user interfaces highly optimized for their exact workflows.

d-buchmann, 2023-03-27:
Exactly, and even integration into ERP is not a pain in the rear end anymore.
But I also see that a full-fledged API is one of the last points on the list, as it must cover all features of the (not yet finished) order and project systems - that's why we're trying out the tablet thing. After all, the PartDB UI is quite nice ;)

jbtronics, 2023-03-27:
I don't think you would have all features to start working on an API. At least for basic part and data structure management it should be relatively easy to implement an API using [APIPlatform](https://api-platform.com/).

Davis8483, 2025-04-17:
@jbtronics on mobile viewing, stocks require scrolling over to click any of the buttons. Would it be possible to instead shift the buttons to be below

[image]

Davis8483, 2025-04-17:
The attachments seem to be formatted oddly as well. The attachments page additionally allows you to scroll down way further than the bottom of the form.

[image]