Part-DB/Part-DB-server #101: Import Parts

mrWheel, 2021-05-03:
It would be nice to have an import function for parts, like the previous version of this great project had (only that one did not work within a docker container .. so I switched to this one).

Or am I missing something?

Pyromane, 2021-05-16:
Import function is currently not available, see: 
https://github.com/Part-DB/Part-DB-symfony/issues/9#issuecomment-572754683

jbtronics, 2023-03-18:
Part import is now possible (see https://docs.part-db.de/usage/import_export.html).

virtadpt, 2023-03-23:
Imports don't work.

Pyromane, 2023-03-23:
> Imports don't work.

Is there a little more detail? Error message?
Can you provide your import file?

virtadpt, 2023-03-23:
Sorry, I was working on updating my comment to include more information.  It's one of those days, you know?  :)

What I'm trying to do: I (finally) got JSON dumps of categories, manufacturers, et al from one of my Part-DB instances.  I was able to import those JSON dumps into a new install.  Stuff that I was able to import: Categories, Manufacturers, Storage, and Suppliers.  I do not have any particular user or group customizations because I'm the only user.  I am currently logged in as admin on the new instance.

I ran JSON dumps of the components already in my old instance.  Export type was "full."  everything fit into seven separate JSON documents.

I am currently logged in as "admin" of the new install.  I am trying to import the first of the parts-dumps.  Tools -> Tools -> Import parts.  It doesn't matter if I let Part-DB infer the file type from the file system or specifically specify JSON, the result is the same.  No "category override," no "needs review" selected.  Whether or not "create unknown data structures" is checked, the result is the same.  Whether or not "abort on invalid data" is checked, the result is the same.

The message displayed is "Content missing."  No parts are imported.  Looking in var/log/prod.log I see the following for every component in the JSON document (slightly reformatted for legibility):

```
{"message":"Uncaught PHP Exception Doctrine\\ORM\\ORMInvalidArgumentException:
\"Multiple non-persisted new entities were found through the given association graph:\n\n
* A new entity was found through the relationship 'App\\Entity\\Parts\\Part#category'
that was not configured to cascade persist operations for entity: Electronic Components. To
solve this issue: Either explicitly call EntityManager#persist() on this unknown entity or
configure cascade persist this association in the mapping for example
@ManyToOne(..,cascade={\"persist\"}).
```
Currently running commit master/0550c04, updated a couple of hours ago.

virtadpt, 2023-03-23:
Here's the import file I'm trying to get loaded.

https://drwho.virtadpt.net/files/part-db-1.json

virtadpt, 2023-03-23:
Another point: If I start over with a brand new, empty database, run `php-8.1 bin/console doctrine:migrations:migrate`, change the admin user's password, and try importing that parts dump I get the same "Content missing" error.

dryas, 2023-03-26:
I have the same issue. I get the message "Content missing" after the upload of the file. I run the official Docker image.

jbtronics, 2023-03-28:
> I have the same issue. I get the message "Content missing" after the upload of the file. I run the official Docker image.

This means that some internal error happens. Normally there should appear a popup, but there was some change in Hotwire Turbo and the error Handling mechanism has changed somehow. I thought I had fixed that, but there still seems to be some edge cases where the popup doesn't appear.

To the import issue: I would need the detailed exception from the error log. Either from `docker log` or the `var/prod/error.log` file.

virtadpt, 2023-03-28:
I just went through the .env and my .env.local files and I don't see any way to create a specific error.log file (nor inside the application's admin pages itself), so here is a link to everything emitted to `var/log/prod.log` during an attempt to import that same file: 

https://drwho.virtadpt.net/files/part-db-server-log-20230328.txt

Note: "Abort on invalid data" is unchecked.  I have also not updated my install since I started commenting on this ticket.

jbtronics, 2023-04-01:
With the latest commits in the master branch, the import with these kinds of nested files should work now. The error popup should now appear correctly, when a server error occurs.

virtadpt, 2023-04-02:
Cool.  I'll give it a try this week (starting a new job tomorrow) and get back to you.

virtadpt, 2023-04-07:
It seems to have worked.  I was able to import my other instance (in the form of JSON files) into a brand-new install.

closed 2023-04-11 by jbtronics