Part-DB/Part-DB-server #109: Typo on page tools/system/user/password - User >disbaled< (no login possible) - should be >disabled<

microplayer, 2021-05-24:
Just a typo in the most recent Docker Image containing Version: 1.0.0-alpha.

[image]

Regards,
Microplayer

microplayer, 2021-06-06:
Text liegt im Repository unter "translations/messages.en.xlf" - dort in Zeile 7541.

Pyromane, 2021-06-06:
Translations are managed by Crowdin, but I can't find a "Part-DB-symfony" project there.
See possibly https://github.com/Part-DB/Part-DB-symfony/pull/83

microplayer, 2021-06-07:
Hi Pyromane,
I'm glad that you gave response to my observation.

Unfortunately I'm not familiar with all those developer tools like Crowdin.
So I have no clue as what to do with that merge(?) #83 from(?) Crowdin...

I did some reading but can't find any obvious reference of a Crowdin integration in the sources files in this repository.
Using the search tool in VSC at least I could identify the line 7541 in file "translations/messages.en.xlf" as the location of that typo.

Pyromane, could you please be so kind to do this simple change using the Crowdin toolchain you brought to the table?
This way we will get rid of it next time when Jan publishes a new docker file to Docker Hub, I assume.

Thank you, Microplayer

coeka, 2021-06-12:
This is the link https://part-db.crowdin.com/part-db

I changed the translation there. Jan just needs to merge pull request and everything will be fine.

jbtronics, 2022-07-21:
Should be fixed now.

closed 2022-07-21 by jbtronics