Part-DB/Part-DB-server #117: Parts withdrawal field within details page

marklinmax, 2021-07-04:
It would be nice if we could withdraw parts from within the parts' details page, just like with the legacy version.

As a part can now have several stocks on different locations, it would probably be good to have a numeric field with a "withdraw" button near it for every location, as those are already shown at the bottom of the details page.

The current way to go to withdraw parts from stock is to edit the part's quantity for a specific location, which is not as straight forward as directly requesting the withdrawal of the specified quantity and let the app calculate the remaining one. It also asks for more unnecessary clicks as one have to enter each part's edit mode.

Maybe there is another way to withdraw parts, but since I couldn't find one, it is likely to not be user-friendly.

sebbalint, 2021-11-23:
I feel the same way and would support the implementation of a withdrawal input field aswell.

jbtronics, 2023-01-08:
Is now implemented. See issue #201

closed 2023-01-08 by jbtronics