Part-DB/Part-DB-server #121: Reverse proxy not working

Chris-42, 2021-08-29:
If I set up a reverse proxy on a diffrent machine in front of the docker container on some calls the docker host is direct accessed.

Setup:
Client -> Proxy-host(apache:80) -> docker-host:10080

the most of the requests are served well to proxy, but requests to 
/tree/categories
/tree/devices
/tree/tools
are sent to docker-host:10080.

**Expected behavior**
all requests should use the proxy as host. So please only use relative URLs.

tmulzer, 2021-11-10:
same same .. I have to use a reverse proxy in a DMZ-setup (can't host part-db on the DMZ-host). 
The node-menus on the left don't make it to the internet.
Client -> DMZ-Host(fqdn:443) -> internal_ip:443

If I call the internal ip, everything is okay & works fine.

kjkriegel, 2021-11-22:
Setting the `TRUSTED_PROXIES` environment to the address of the reverse proxy appears to fix this.

jbtronics, 2022-07-17:
See issue #145 .
Setting `TRUSTED_PROXIES` env variable should help.

closed 2022-07-17 by jbtronics