Part-DB/Part-DB-server #128: LDAP Support for user authentification in Part-DB

sebbalint, 2021-11-23:
**Is your feature request related to a problem? Please describe.**
In our group we already have a functioning LDAP. We currently need to log in seperatly to the Part-DB.

**Describe the solution you'd like**
 To avoid unnecessary log-ins to the database it would be nice to access the database with the already in use LDAP. Therefore LDAP support would be a huge enhancement for our user experience and ease working with the Part-DB.

tmulzer, 2021-12-11:
Same in our environment .. our "LDAP" in fact is an Active Directory...
Another (appreciated) option would be to use an intermediary keycloak.

lucalutz2005-dev, 2022-04-25:
> Same in our environment .. our "LDAP" in fact is an Active Directory... Another (appreciated) option would be to use an intermediary keycloak.

+1 for Keycloak, OpenID - Single Sign-on would be a huge benefit

jbtronics, 2022-07-18:
It is one of my (lower priority) goals to support LDAP authentication in Part-DB.
I dont have much knowledge of LDAP, but i see a bit of a problem with the permissions system of Part-DB and how to map which LDAP user can do what.

lucalutz2005-dev, 2022-07-20:
I highly reccomnend you to take a look at OpenID2. Its much more powerful than LDAP but with keycloak you can run an LDAP server as user backend. But it handels single sign on and in my opinion is the future. A lot of (OpenSource) projects already change from LDAP to OpenID2

lucalutz2005-dev, 2022-07-20:
In LDAP as well as in OpenID2 setups groups are often used for the permissions management

jbtronics, 2022-07-20:
The problem is that with the local user system permissions are not only group based, but user based (you can change the permissions for each user separately).

But In Principle we could just map an LDAP field (or user group) to an Part-DB group.

lucalutz2005-dev, 2022-07-20:
Sounds good, LDAP with keycloak shouldnt be to bad to implement

jbtronics, 2023-02-21:
I have done some experiments with SAML integration in Part-DB and it works pretty well and allows connection to Keycloak. SAML is a bit more verbose to configure (as you have several certificates) than OpenID Connect, but for SAML there are better libraries for integration and in the end it should not matter much as most identity provider servers support both. This should allow to use Keycloak (and therefore all backends keycloak supports) with Part-DB.

When a user logins for the first time via SSO a user account is created in Part-DBs database, which holds configurations and so on. The userinfo (email, firstname, lastname, department) is filled by the attributes supplied from the IdP (if existing) and updated on every login. 

For permission handling, I plan that you can configure a ID of the (Part-DB) user group, to which SSO users gets assigned to on their first login, either by a global configured one or (optionally) per user via an SAML attribute, which the IdP supplies.
If you need some more control an Part-DB administrator can still change the permissions of a SSO user like normally.

This is not perfect, but should be sufficient for most usecases I guess (where you have many users with the same permissions) and very few special ones. Using datamapping in keycloak you should be even able to map users to Part-DB groups based on their keycloak roles / groups.

jbtronics, 2023-03-04:
Authentication support for SAML (for use with Keycloak) is now working and can be found in the master-branch. It will be part of the upcoming 1.1.0 release.

You can find the documentation for setting up SAML here: https://docs.part-db.de/installation/saml_sso.html