Part-DB/Part-DB-server #134: Maybe an error in the description Part-DB-symfony/docs/docker/docker-install.md

christiannolte, 2022-01-17:
**Describe the bug**
Following the steps https://github.com/Part-DB/Part-DB-symfony/blob/master/docs/docker/docker-install.md#direct-use-of-docker-image cause a problem

**To Reproduce**
Steps to reproduce the behavior:
1. create a directory media for the volume
2. create a directory uploads for the folume
2. Start the image jbtronics/part-db1:master with the two mentioned volumes given to the container
2. Execute Step 4 like mentioned
3. This will result in an error

**Expected behavior**
A database should be created/migrated.

**Additional context**
When not using volumes everything is fine. So using volumes is overwriting/overlaying e.g. the database app.db.
Maybe there should be a step to create the database in the volume.

Thanks
   Christian

christiannolte, 2022-01-17:
Ich habe versucht mit 
docker cp container:/var/www/html/uploads/app.db
in mein Volume zu kopieren und dann das Volume zu übergeben, das hat aber leider nicht geklappt.

jbtronics, 2022-07-17:
This is indeed an issue with the mapping of the database file. By default, the db lives inside the var/ folder which is not mapped. 
You have to put it inside a mapped folder via the `DATABASE_URL` env. I added some hint about this to the docker-install.md and updated the docker-compose.yaml example to use this.

closed 2022-07-17 by jbtronics