Part-DB/Part-DB-server #135: SQL Error while running migration script

Joshua265, 2022-01-20:
**Describe the bug**
while running the migration script the following error occurs. Even after running the migration script, the partdb frontend responds with an error 500.

**To Reproduce**
Steps to reproduce the behavior:
1. build and run docker image
2. connect it to mariadb docker container as database
3. have data of previous partdb in mariadb
4. sudo docker exec -it partdb sh  
5. php bin/console doctrine:migrations:migrate

**Expected behavior**
script runs without errors

**console output**
[notice] Migrating up to DoctrineMigrations\Version20200502161750                                                                                                                                             [error] Migration DoctrineMigrations\Version20200126191823 failed during Execution. Error: "An exception occurred while executing 'DROP INDEX id_user ON log':                                                                                                                                                                                                                                                              SQLSTATE[42000]: Syntax error or access violation: 1091 Can't DROP INDEX `id_user`; check that it exists"                                                                                                     16:25:15 CRITICAL  [console] Error thrown while running command "doctrine:migrations:migrate". Message: "An exception occurred while executing 'DROP INDEX id_user ON log':\n\nSQLSTATE[42000]: Syntax error or access violation: 1091 Can't DROP INDEX `id_user`; check that it exists" ["exception" => Doctrine\DBAL\Exception\DriverException^ { …},"command" => "doctrine:migrations:migrate","message" => """  An exception occurred while executing 'DROP INDEX id_user ON log':\n  \n  SQLSTATE[42000]: Syntax error or access violation: 1091 Can't DROP INDEX `id_user`; check that it exists  """]                                                                                                                                                                                                                                            In AbstractMySQLDriver.php line 128:                                                                                                                                                                                                                                                                                                                                                                                          An exception occurred while executing 'DROP INDEX id_user ON log':                                                                                                                                                                                                                                                                                                                                                          SQLSTATE[42000]: Syntax error or access violation: 1091 Can't DROP INDEX `id_user`; check that it exists                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                In Exception.php line 18:                                                                                                                                                                                                                                                                                                                                                                                                     SQLSTATE[42000]: Syntax error or access violation: 1091 Can't DROP INDEX `id_user`; check that it exists                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                In PDOConnection.php line 141:                                                                                                                                                                                                                                                                                                                                                                                                SQLSTATE[42000]: Syntax error or access violation: 1091 Can't DROP INDEX `id_user`; check that it exists                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                doctrine:migrations:migrate [--write-sql [WRITE-SQL]] [--dry-run] [--query-time] [--allow-no-migration] [--all-or-nothing [ALL-OR-NOTHING]] [--configuration CONFIGURATION] [--em EM] [--conn CONN] [--] [<version>]    

**Server Side**
- Part-DB Version: https://github.com/Part-DB/Part-DB-symfony/commit/e942ab1bf6e6de0c216862b2b4b58f1375a27ca8
- PHP Version: PHP 7.4.27 (cli) (built: Dec 21 2021 21:31:45) ( NTS ) 
- Database Server: latest mariadb docker image

Xernito, 2023-02-14:
For the mariadb it should by fixed by change the .env.local:

add: ?serverVersion=mariadb-10.5.0 <- your DB version

But i got the same error with MySQL DB Version 5.7

jbtronics, 2023-02-14:
> For the mariadb it should by fixed by change the .env.local:
> 
> add: ?serverVersion=mariadb-10.5.0 <- your DB version
> 
> But i got the same error with MySQL DB Version 5.7

The serverVersion should not change the behavior of Part-DB in normal mode, and especially not for migrations (as they are just plain SQL executed).

This is most likely an issue in your current database (or an incompatibility of the migration with your database). It is important that you do the migration on a clean state. If the database migration has failed, you have to restore your database from the backup, before doing the migration again (or it will just fail again and give you wrong error messages)

To debug this I need the info whether the database was migrated from a legacy Part-DB and in the best case an (anonymized) dump of your database.

Xernito, 2023-03-11:
Hello,

I used a fresh and empty database. With no data in it or even an old version of Part-DB.
`Server-Version: 5.7.41-0ubuntu0.18.04.1 - (Ubuntu)
PHP-Version: 7.2.24-0ubuntu0.18.04.17
Datenbank-Client Version: libmysql - mysqlnd 5.0.12-dev - 20150407`

Now I used Sqlite and the migration works fine.

I think it's a problem between docrine an MySQL Server Version.

closed 2023-04-23 by jbtronics