Part-DB/Part-DB-server #136: Error 50, can't upload files to footprints

Protoncek, 2022-01-30:
**Describe the bug**
When creating a footprint and i try to upload an image of it from my local PC (via "browse") uploads ends with an "internal server error", code 500.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Eidt --Footprints
2. Click on attachment
3. Click on "Browse" and select an image
4. Click "save"

**Expected behavior**
An uploaded image of footprint

**Screenshots**
[image]

**Server Side**
- Part-DB Version: [e.g. Part-DB V1.0.o-aplha
- PHP Version: 7.0,7.3,7.4, all installed on Synology
- Database Server MariaDB10 on Synology

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser Edge
 - Version latest

Protoncek, 2022-01-30:
Sorry, it's an Error 500, not 50... i can't edit title...

Xernito, 2022-03-15:
Hi,

what image did you use? I didn't get any error

nskartem, 2022-05-15:
@Xernito I've just tried to use both master and nightly images, and still have that error on uploading file:(
`Unable to open image /var/www/html/public/media/part/foto-6280e5ebbe081.jpg`
I've checked all the paths and permissions, and I can see uploaded image on <IP>/media/part/foto-6280e5ebbe081.jpg, so I think, it's something in the code.

nskartem, 2022-05-15:
@jbtronics, hello!
I think, that issue is bounded with issue #122 and with issue #115.
I've configured PHP-GD to support Jpeg, and issue was solved. I see a lot of differences between Dockerfile in the repo and image on Dockerhub: https://hub.docker.com/layers/part-db1/jbtronics/part-db1/master/images/sha256-b0f144ae4a7f8412a8525e509875dade70ed8f786ca7e427ded06b571c2f049d?context=explore
Could you please correct your main Dockerhub image to support JPG?

jbtronics, 2022-07-21:
The docker image now contains a GD with jpeg and webp support, so the problem should be fixed.
Also if something fails during creation of thumbnail images, an exception is no longer thrown, but the original file is shown as fall back.

closed 2022-07-21 by jbtronics