Part-DB/Part-DB-server #145: Part-DB tries to access HTTP Content

lucalutz2005-dev, 2022-05-04:
**Describe the bug**
When setting up the docker-compose setup with ngninx reverse-proxy the browser throws a mixed content error becouse Part-DB tries to access "http://inv.domain/en/tree/categories", "http://inv.domain/en/tree/devices" and "http://inv.domain/en/tree/tools". The problem is this is happening if you access https://inv.domain/en/login. My nginx-configis the following:

```conf
server {
    listen 80;
    listen 443 ssl;
    server_name inv.domain;

    ssl_certificate        /etc/nginx/certs/live/domain/fullchain.pem;
    ssl_certificate_key    /etc/nginx/certs/live/domain/privkey.pem;

    client_max_body_size 500M;

    location / {
        proxy_pass http://IP:8085/;
        proxy_http_version 1.1;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection "Upgrade";
        proxy_set_header Host $host;
        proxy_set_header X-Forwarded-Host $host;
        proxy_set_header X-Forwarded-Proto https;
        proxy_set_header X-Forwarded-Server $host;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
    }

    if ($scheme = http) {
      return 301 https://$server_name$request_uri;
    }
}
```

**To Reproduce**
Steps to reproduce the behavior:
Use the described docker-compose file (uses the image with the :master tag) and the Nginx-config above

**Expected behavior**
HTTPS requests, becouse the reverse proxy should tell Part-DB the protocol

Thank you already

jbtronics, 2022-07-17:
For security reasons the forward-headers are normally ignored, until the proxy origin is marked as trusted.

In the new version you can set the trusted proxies via `TRUSTED_PROXIES` environment variable (by default just `127.0.0.1` is set as trusted).
So if the reverse proxy is on the same IP as your Part-DB it should work now. Otherwise you have to set the `TRUSTED_PROXIES` option (via docker config) to match the IP of your reverse proxy.

lucalutz2005-dev, 2022-07-20:
Thank you very much, that helps 👍

closed 2022-07-20 by lucalutz2005-dev