Part-DB/Part-DB-server #146: 405 HTTP error on part deletion.

nskartem, 2022-05-16:
When I'm trying to delete part, I'm getting the "No route found for "POST http://168.138.69.59/en/part/6/delete": Method Not Allowed (Allow: DELETE)" error.

Steps to reproduce the behavior:
1. Try to delete any part from */part/*/info
2. Get error screen with: "No route found for "POST http://168.138.69.59/en/part/6/delete": Method Not Allowed (Allow: DELETE)"

**Expected behavior**
Part deleted.

**Server Side**
- Part-DB Version: master image from Dockerhub (hash sha256:b0f144ae4a7f8412a8525e509875dade70ed8f786ca7e427ded06b571c2f049d)
- PHP Version: 7.4.29
- Database Server: SQLite

**Additional context**
Maybe it's bounded with issue #136, and it's broken after I tried jpg upload, but I deleted all the jpg attachments and restarted the container, so I don't think so. Possible apache2 configuration bug?

jbtronics, 2022-07-21:
Should be fixed now in the latest version.

virtadpt, 2022-07-21:
I just updated my install to the latest Git commit, and it's still returning HTTP 405's.

jbtronics, 2022-07-21:
Have you installed the new composer dependencies and done a `php bin/console cache:clear` after pulling the latest git commit?

nskartem, 2022-07-22:
Hello, @jbtronics!

Thanks for the fix, I'm using Docker image, which is built from this repo's Dockerfile (I'm running PartDB on a aarch64 server now, so I can't use dockerhub images). I've done git pull from master and rebuilt image, now deletion is working right:
[image]
Thank you!

jbtronics, 2022-07-22:
Maybe as a sidenote: In Principle the GitHub action i use to automatically build the Docker images can build images for other platform too (including several ARM architectures).

The problem is that as these other architectures has to be emulated via QEMU, this is very slow (and not reasonable to do for every commit). When there will be tagged  releases of Part-DB, however I plan to include images for other architectures than amd64 too.

nskartem, 2022-07-22:
By the way, if you need some help with automation/build/pipelines/etc. (without coding), I'd like to help you, because it'll be interesting to support such a great open source project. I work in a middle-sized company as an outsource Linux administrator/DevOps.

virtadpt, 2022-07-22:
@jbtronics Yes, I have.

virtadpt, 2022-08-01:
Updated to master/a8bd894.  Able to delete parts now but that's about it.

jbtronics, 2023-02-06:
Should be fixed some time ago.

closed 2023-02-06 by jbtronics