Part-DB/Part-DB-server #147: Part-DB-symfony throws 302 errors after heavy use (with solution)

virtadpt, 2022-06-10:
**Describe the bug**
After sweating a couple of hundred inventory entries into my Part-DB-symfony it started throwing HTTP 302 errors rather than log me in.  Using my browser's debugging tools I was able to confirm what was going on.  I did come up with a fix but it might need a patch to the software itself.

**To Reproduce**
Steps to reproduce the behavior:
1. Set up Part-DB-symfony normally.
2. Log in.
3. Add a large number of parts to your inventory.
4. Log out (or close your browser).
5. Try to log back in.
6. Get a blank screen for a while, then browser reports too many 302 redirects.

**Expected behavior**
I get the Part-DB-symfony login screen, log in, and use the webapp normally.

**Screenshots**
n/a

**Server Side**
- Part-DB Version: 3e0745f859d17389b8c652bd39f69401410be645
- PHP Version: v8.2
- Database Server: MySQL 5.7.29

**Desktop (please complete the following information):**
 - OS: OSX
 - Browser: Chrome
 - Version: 102.0.5005.115 (Official Build)

**Additional context**
The fix involved deleting the cookies for my Part-DB-symfony installation and forcing a refresh of the page.  Then I could log back in.  I don't know if this is a bug in Part-DB-symfony or not.

virtadpt, 2022-06-10:
I noticed something: Looking at the cookies, one of them was set for .inventory.example.com (a dot at the very front) and the other (the PHPSESSID cookie) was set to inventory.example.com (no dot at the beginning).  Could that have something to do with it?

jbtronics, 2022-07-23:
This could maybe related to some issue in the frontend (ajax) code. It sometimes behaves a bit strange, especially if you start many requests simultaneously. I have started to rework my self written frontend code to [Hotwire Turbo](https://turbo.hotwired.dev/) (see `turbo` branch), which is much more stable (and offers a better user experience) than my code.
Maybe the issue is gone then.

virtadpt, 2022-07-24:
Hmm... I'll give it a try tonight.  Are there any migrations or downgrades we should be aware of?

jbtronics, 2022-07-24:
The code in the `turbo` branch is not recommended for any productions and still has some issues, as i still trying to get everything to work with the new frontend system.

virtadpt, 2022-07-24:
Good to know - thank you.

virtadpt, 2022-08-01:
This hasn't been an issue for a couple of updates.  Feel free to close it out.

closed 2022-08-01 by jbtronics