Part-DB/Part-DB-server #148: Clicking on "Save" buttons always shows a "Failed!" pop-up.

virtadpt, 2022-06-10:
**Describe the bug**
Whenever saving my work, be it editing a user account or entering a new part, when click the Save button I always get a little "Failed!" pop-up bubble on the button, even though it saved correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. Log into your Part-DB-symfony install.
2. Go to any major feature, for example, System->Users->Edit user
3. Select a user account to edit.
4. Make a trivial change, or no change at all.
5. Click the "Save" button.
6. Get a "Failed!" bubble.

**Expected behavior**
There should be no "Failed!" bubble for things that worked.

**Screenshots**
Disappears before I can get a screenshot.  Sorry.

**Server Side**
- Part-DB Version: 3e0745f859d17389b8c652bd39f69401410be645
- PHP Version: v8.1
- Database Server: MySQL 5.7.29

**Desktop (please complete the following information):**
 - OS: OSX
 - Browser: Chrome
 - Version: 102.0.5005.115 (Official Build)

**Additional context**
This happens for the built-in admin user as well as any user accounts I create without admin privileges.

jbtronics, 2022-07-20:
Should be fixed now.

closed 2022-07-20 by jbtronics

virtadpt, 2022-07-21:
Thanks - I'll try it this weekend.

virtadpt, 2022-07-21:
I just updated my install to the latest Git commit, and it seems to work now.  Thanks!