Part-DB/Part-DB-server #149: Unable to delete Devices.

virtadpt, 2022-06-10:
**Describe the bug**
Creating a Device (as the admin, Tools->Edit->Device) works as expected.  Attempting to delete a Device does not.

**To Reproduce**
Steps to reproduce the behavior:
1. Log into your Part-DB-symfony install as the admin user.
2. Tools->Edit->Device
3. Create a new device, e.g., "Digital VOM".
4. Click "Create element".
5. On the same screen, click the device to edit it.
6. Click "Delete element".
7. Click "OK" when asked if you really want to delete the device.
8. Wait. The device is still in the list of devices on the "Edit device" page.  The device is still present in the Device->Root node list.
9. Log out.  Force a refresh.  Log back in.  It's still there.
10. Log in with a totally different browser.  It's still there.

**Expected behavior**
Deleting a device should make it go away.

**Screenshots**
Before: 
[image]

During:
[image]

After:
[image]

**Server Side**
- Part-DB Version: 3e0745f859d17389b8c652bd39f69401410be645
- PHP Version: PHP v8.1
- Database Server: MySQL v5.7.29

**Desktop (please complete the following information):**
 - OS: OSX
 - Browser: Chrome
 - Version: 102.0.5005.115 (Official Build)

**Additional context**
This happens reliably and repeatably when logged in as the admin user.

jbtronics, 2022-07-21:
Devices are not really implemented yet. But the issue you found is related to some bigger issue, where deletion of almost any element was possible. Now things should work as intended.

closed 2022-07-21 by jbtronics

virtadpt, 2022-07-21:
I just updated my install to the latest Git commit and tried deleting a device.  It still doesn't work.

virtadpt, 2022-07-22:
After a cache clear, trying to delete anything in my inventory I still get an HTTP 405 error popup.

virtadpt, 2022-08-01:
Updated my install to master/a8bd894 and tried to delete a device.  It still doesn't work.

Test 1:
* Part-DB frontpage
* Sidebar: Devices -> Root node -> pick a device (e.g., Audio Equipment)
* Part-DB page does not change (device highlights as selected but page does not update).

Test 2:
* Part-DB frontpage
* Sidebar: Tools -> Edit -> Devices
* Device management page appears
* Pick a device (e.g., Audio Equipment)
* Delete element
* Confirm in the popup.
* "Element still contains parts! You have to move the parts, to be able to delete this element."
* Export -> JSON, Extended, Include children elements in export
* Download, look at contents of exported JSON document:

```
[me @ laptop:() Downloads]$ python3 -m json.tool export_Audio\ Equipment_extended.json
[
    {
        "children": [],
        "comment": "",
        "name": "Audio Equipment",
        "lastModified": "2022-06-08T23:48:54+00:00",
        "addedDate": "2022-06-08T23:48:54+00:00",
        "type": null
    }
]
```

Even created a brand-new Test Device with absolutely nothing associated with it.  Also doesn't work.

virtadpt, 2022-08-03:
Just re-tried with "Delete recursive (all subelements)" checked.  Also does not work.

virtadpt, 2022-09-07:
Just updated to commit 8e6bc25d2e84beba314e073714d82a247936f3ed.  This function still doesn't work.

virtadpt, 2022-12-27:
Welp, I bit the bullet and hand-hacked the database to get rid of the devices.  I had to clear and warm up the on-disk caches to finish the job but that did the trick.  It was just a matter of deleting rows from a table.

jbtronics, 2022-12-30:
For the devices the remove function was deactivated intentionally, as you could not view the contained parts, which could have caused data loss from old part-db versions (and other problems).

I am currently working on some basic device (now project system), then everything is easier.

virtadpt, 2022-12-31:
That's good to know, thank you.  When I see it pop up I'll poke at it a little.