Part-DB/Part-DB-server #154: How to change currency with docker?

jagauthier, 2022-07-29:
I've installed the docker image. (I had to add mysql support)

Under my user the preferred currency is Euro.  I saw this tidbit:

```
php bin/console app:update-exchange-rates: 
```
Update the exchange rates of your currencies from internet. Setup this to be run in a cronjob to always get up-to-date exchange rates. If you dont use Euro as base currency, you have to setup an fixer.io API key in ```.env.local```

There isn't any .env.local, and I don't see any references to fixer.io in the existing .env file.

Running the command mentioned doesn't return anything useful:
```
 ! [NOTE] Update currency exchange rates with base currency: EUR

 [OK] 0 (of 0) currency exchange rates were updated.
```

jagauthier, 2022-07-29:
Okay, so maybe those instructions are not quite right?
I added a new currency. USD.
But 
```
[WARNING] The exchange query "USD/EUR" is not supported by the service "Exchanger\Service\Chain".
```

When running the above command.

jagauthier, 2022-08-01:
So, it looks like this is actually done in parameters.yaml?
I changed that (and created a volume map out to it) and it seems to work now.

jbtronics, 2022-12-11:
Now it should work without any additional configuration, at least if your base currency is euro.
For retrieving exchange rates with euros as base, it is configured to use the API of the [european central bank](https://www.ecb.europa.eu/stats/policy_and_exchange_rates/euro_reference_exchange_rates/html/index.en.html) to fetch exchange rates of euros into other common currencies, which can be used without any API keys.

If you have some other base currencies (like USD) or need some more exotic exchange rates, you need to configure a provider like fixer.io, which needs an API key.

Before there were in issue in the exchange rate logic, which caused to use the wrong base currency, so the ECB provider were not used, and you always had to configure an additional provider. Now everything should work out of the box, without the need for configuration (at least for the default euro base currency). And you can use the web UI and the console command to automatically update the exchange rates.

closed 2022-12-11 by jbtronics