Part-DB/Part-DB-server #159: Attach files other than footprints.

virtadpt, 2022-08-04:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
As a user who frequently purchases PCBs from online manufacturers, and who maintains an inventory of those PCBs in Part-DB, it would be very helpful if I could upload the .zip files of the Gerber files and attach them to their records.

**Describe alternatives you've considered**
PCB Way already makes it possible to download archives of uploaded Gerber files, et al, but they can go away at any time and I don't always have my personal laptop handy to grab my local copies.

jbtronics, 2022-11-10:
I do not quite understand your problem. You can upload any file type into an attachment for any element that can have attachments (like parts, categories, footprints, etc...).

You can restrict which file types are allowed for an attachment type (e.g. you can define an datasheet attachment type which only allows PDF uploads), but by default you can upload everything, as long as it does not exceeds the upload limit of your website (but you can configure this in your server config).

virtadpt, 2022-11-10:
I tried to upload the .zip files and got an error message.  It may have been due to .zip not being defined as an uploadable file type.

Now I just have to figure out why my install's broken.

closed 2023-01-22 by jbtronics