Part-DB/Part-DB-server #167: Make the ID editable or add IPN (Internal Part Number)

zanna-37, 2022-08-09:
**Is your feature request related to a problem? Please describe.**
I keep track of items with an internal part number which should be modifiable and searchable.

**Describe the solution you'd like**
It would be amazing if the ID could be manually modified. Alternatively, a unique indexed field (e.g. IPN - Internal Part Number) would work.

**Describe alternatives you've considered**
Stick with the generated ID =(

jbtronics, 2022-08-09:
The database ID is used to identify the relations between the different entities (and is even create by the database server itself when a new entry is created). When you can change it, this will most breaking the numbering system and relationship between entities. Also the ID value is restricted to a single number, which i guess is not really useful.

An text field representing an user provided  internal ID is possible however (and it will behave like any other field of the part).
 I am interested however what is your use case for this?

jagauthier, 2022-08-09:
@zanna-37  Use the tag for your IPN? It is searchable

zanna-37, 2022-08-09:
Doable, yes, but it isn't guaranteed to be unique. On top, I would generate a ton of tags that make the use of real tags inconvenient because their are overwhelmed by IPN.

jbtronics, 2022-12-05:
There is now an Internal Part number field in the advanced section of a part, which can be used for this usecase.

closed 2022-12-05 by jbtronics

zanna-37, 2022-12-07:
Thank you very much!
However, important characteristics, in my opinion, are that the IPN should be unique and searchable.
Are these two conditions respected?

jbtronics, 2022-12-07:
The IPN are ensured to be unique across all parts. That is ensured by a unique database on database level and a check in the software, which shows an error message if you want to save an already exisiting IPN to another part.
You can search/filter by the IPN via the filter system (there is now a new IPN filter).

The IPN is a generic text field, too be as generic as possible and to support all naming schemes somebody could want.

However this makes it very difficult until impossible to implement a next-available IPN feature, as predicting the next IPN for all possible formats is not really possible (whats the next free number for a IPN like "1234-A-53" or so?).

zanna-37, 2022-12-15:
It will suffice. =)
Thank you very very much for everything.