Part-DB/Part-DB-server #168: Theme Issue after update to latest version

tbnobody, 2022-08-14:
I've just updated my PartDB instance to the latest version and the theme looks a bit strange:
[image]

The search bar is aligned left and the accordion is just a grey button. 

[image]
Also the tools don't look correct. (see Toggle Dropdown)

**To Reproduce**
Steps to reproduce the behavior:
I just got the latest version using the following commands:
```
sudo -u www-data git pull 
sudo -u www-data ./composer.phar install --no-dev -o 
sudo -u www-data yarn install 
sudo -u www-data yarn build 
sudo -u www-data php bin/console cache:warmup 
sudo -u www-data php bin/console doctrine:migrations:migrate
```

**Server Side**
- Part-DB Version: e6408a4dff5b95e9c9fee3336c82a21120c1eb2e
- PHP Version: 7.4.30
- Database Server: MariaDB 10.5.15

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser Vivaldi
 - Version 5.3.2679.68 (Stable channel) (32-Bit) 

Not sure if this error occours because of the ongoing migration to turbo or if I am missing something.

jbtronics, 2022-08-14:
That is mostly likely due to the change from Bootstrap 4 to bootstrap 5, and your browser still uses the old stylesheets.

Clearing the browser cache should help.

tbnobody, 2022-08-14:
Hm you are right. It's working with firefox and chrome but not in vivaldi (even in a private tab)

tbnobody, 2022-08-14:
Ok, sorry for bother you... after deleting the cache its working... (already tried ctrl+f5 etc to force a reload but without success)

closed 2022-08-14 by tbnobody