Part-DB/Part-DB-server #170: Error migrating database

Tropaion, 2022-08-26:
@jbtronics Since the installation worked, I now tried migrating the database, but got this error:
```
root@PartDB:/var/www/partdb# sudo -u www-data php bin/console doctrine:migrations:migrate

 WARNING! You are about to execute a migration in database "partdb" that could result in schema changes and data loss. Are you sure you wish to continue? (yes/no) [yes]:
 > yes

[notice] Migrating up to DoctrineMigrations\Version20200502161750
[warning] Old Part-DB Database detected! Continue with upgrade...
[warning] Migration DoctrineMigrations\Version1 was executed but did not result in any SQL statements.
[warning] [!!!] Permissions were updated! Please check if they fit your expectations!
[error] Migration DoctrineMigrations\Version20190902140506 failed during Execution. Error: "An exception occurred while executing a query: SQLSTATE[23000]: Integrity constraint violation: 1452 Cannot add or update a child row: a foreign key constraint fails (`partdb`.`#sql-alter-11a-1a708`, CONSTRAINT `FK_1483A5E9FE54D947` FOREIGN KEY (`group_id`) REFERENCES `groups` (`id`))"
22:46:58 CRITICAL  [console] Error thrown while running command "doctrine:migrations:migrate". Message: "An exception occurred while executing a query: SQLSTATE[23000]: Integrity constraint violation: 1452 Cannot add or update a child row: a foreign key constraint fails (`partdb`.`#sql-alter-11a-1a708`, CONSTRAINT `FK_1483A5E9FE54D947` FOREIGN KEY (`group_id`) REFERENCES `groups` (`id`))" ["exception" => Doctrine\DBAL\Exception\ForeignKeyConstraintViolationException^ { …},"command" => "doctrine:migrations:migrate","message" => "An exception occurred while executing a query: SQLSTATE[23000]: Integrity constraint violation: 1452 Cannot add or update a child row: a foreign key constraint fails (`partdb`.`#sql-alter-11a-1a708`, CONSTRAINT `FK_1483A5E9FE54D947` FOREIGN KEY (`group_id`) REFERENCES `groups` (`id`))"]

In ExceptionConverter.php line 58:
                                                                                                         
  An exception occurred while executing a query: SQLSTATE[23000]: Integrity constraint violation: 1452   
  Cannot add or update a child row: a foreign key constraint fails (`partdb`.`#sql-alter-11a-1a708`, CO  
  NSTRAINT `FK_1483A5E9FE54D947` FOREIGN KEY (`group_id`) REFERENCES `groups` (`id`))                    
                                                                                                         

In Exception.php line 30:
                                                                                                         
  SQLSTATE[23000]: Integrity constraint violation: 1452 Cannot add or update a child row: a foreign key  
   constraint fails (`partdb`.`#sql-alter-11a-1a708`, CONSTRAINT `FK_1483A5E9FE54D947` FOREIGN KEY (`gr  
  oup_id`) REFERENCES `groups` (`id`))                                                                   
                                                                                                         

In Connection.php line 72:
                                                                                                         
  SQLSTATE[23000]: Integrity constraint violation: 1452 Cannot add or update a child row: a foreign key  
   constraint fails (`partdb`.`#sql-alter-11a-1a708`, CONSTRAINT `FK_1483A5E9FE54D947` FOREIGN KEY (`gr  
  oup_id`) REFERENCES `groups` (`id`))                                                                   
                                                                                                         

doctrine:migrations:migrate [--write-sql [WRITE-SQL]] [--dry-run] [--query-time] [--allow-no-migration] [--all-or-nothing [ALL-OR-NOTHING]] [--configuration CONFIGURATION] [--em EM] [--conn CONN] [--] [<version>]

root@PartDB:/var/www/partdb# 
```
After running it a second time, it says this:
```
root@PartDB:/var/www/partdb# sudo -u www-data php bin/console doctrine:migrations:migrate

 WARNING! You are about to execute a migration in database "partdb" that could result in schema changes and data loss. Are you sure you wish to continue? (yes/no) [yes]:
 > yes

[notice] Migrating up to DoctrineMigrations\Version20200502161750
[error] Migration DoctrineMigrations\Version20190902140506 failed during Execution. Error: "An exception occurred while executing a query: SQLSTATE[42S02]: Base table or view not found: 1146 Table 'partdb.attachement_types' doesn't exist"
22:50:21 CRITICAL  [console] Error thrown while running command "doctrine:migrations:migrate". Message: "An exception occurred while executing a query: SQLSTATE[42S02]: Base table or view not found: 1146 Table 'partdb.attachement_types' doesn't exist" ["exception" => Doctrine\DBAL\Exception\TableNotFoundException^ { …},"command" => "doctrine:migrations:migrate","message" => "An exception occurred while executing a query: SQLSTATE[42S02]: Base table or view not found: 1146 Table 'partdb.attachement_types' doesn't exist"]

In ExceptionConverter.php line 51:
                                                                                                                                                     
  An exception occurred while executing a query: SQLSTATE[42S02]: Base table or view not found: 1146 Table 'partdb.attachement_types' doesn't exist  
                                                                                                                                                     

In Exception.php line 30:
                                                                                                      
  SQLSTATE[42S02]: Base table or view not found: 1146 Table 'partdb.attachement_types' doesn't exist  
                                                                                                      

In Connection.php line 72:
                                                                                                      
  SQLSTATE[42S02]: Base table or view not found: 1146 Table 'partdb.attachement_types' doesn't exist  
                                                                                                      

doctrine:migrations:migrate [--write-sql [WRITE-SQL]] [--dry-run] [--query-time] [--allow-no-migration] [--all-or-nothing [ALL-OR-NOTHING]] [--configuration CONFIGURATION] [--em EM] [--conn CONN] [--] [<version>]

root@PartDB:/var/www/partdb# 
```

Do you know what this is?

coeka, 2022-08-30:
I had the same issue: https://github.com/Part-DB/Part-DB-symfony/issues/67
In my case one of my users had the group_id 0. I imported my backup, changed the group_id and startet again the migration.

jbtronics, 2022-09-01:
@Tropaion What DB server are you using (MySQL or MariaDB? What version? There are some subtle difference between both versions). From which (old) Part-DB are you updating?

That the migration fails at the second time after an error happened in first attempt is normal (unfortunatly), as MySQL does not support transactions for schema changes, so an failing migration is applied partialy to the DB, which makes it very difficult to continue...

Tropaion, 2022-09-01:
@jbtronics Thanks for the answer.
I'm using MariaDB Server-Version: 10.5.15-MariaDB-0+deb11u1 - Debian 11 and my old PartDB is v0.6.0-dev.
My partdb: parts.flowerhouse.at
Database (maybe it helps?):  [partdb.zip](https://github.com/Part-DB/Part-DB-symfony/files/9470324/partdb.zip)

jbtronics, 2022-09-17:
So these error messages were caused by multiple errors:
* Old Part-DB somehow set 0 as a group_id, which is not allowed now (I guess this happened, when a user did not have a group assigned)
* When a part were deleted, the relating order and price details (and maybe the attachments too) were not deleted automatically (unlike now), which caused problems with the new foreign key constraints.

The database now gets cleaned before the real migration starts, so the migration from your database should work now with the latest Part-DB version.

closed 2022-09-17 by jbtronics

Tropaion, 2022-09-18:
Thanks, I tried it, migrating works now.