Part-DB/Part-DB-server #172: Error: No extension able to load the configuration for "fos_ck_editor"

virtadpt, 2022-09-05:
**Describe the bug**
I've just updated my instance to commit 8e6bc25d2e84beba314e073714d82a247936f3ed.  When running the command `php-8.1 bin/console cache:clear` the following error message is thrown:

```
 // Clearing the cache for the prod environment with debug                     
 // false                                                                      

In FileLoader.php line 174:
                                                                              
  There is no extension able to load the configuration for "fos_ck_editor" (  
  in "/home/virtadpt/merivis.virtadpt.net/Part-DB-symfony/config/packages/fo  
  s_ckeditor.yaml"). Looked for namespace "fos_ck_editor", found ""framework  
  ", "sensio_framework_extra", "doctrine", "doctrine_migrations", "security"  
  , "twig", "monolog", "webpack_encore", "datatables", "shivas_versioning",   
  "liip_imagine", "twig_extra", "gregwar_captcha", "translation", "florianv_  
  swap", "nelmio_security", "turbo", "scheb_two_factor", "ru2_f_two_factor""  
   in /home/virtadpt/merivis.virtadpt.net/Part-DB-symfony/config/packages/fo  
  s_ckeditor.yaml (which is being imported from "/home/virtadpt/merivis.virt  
  adpt.net/Part-DB-symfony/src/Kernel.php").                                  
                                                                              

In YamlFileLoader.php line 799:
                                                                              
  There is no extension able to load the configuration for "fos_ck_editor" (  
  in "/home/virtadpt/merivis.virtadpt.net/Part-DB-symfony/config/packages/fo  
  s_ckeditor.yaml"). Looked for namespace "fos_ck_editor", found ""framework  
  ", "sensio_framework_extra", "doctrine", "doctrine_migrations", "security"  
  , "twig", "monolog", "webpack_encore", "datatables", "shivas_versioning",   
  "liip_imagine", "twig_extra", "gregwar_captcha", "translation", "florianv_  
  swap", "nelmio_security", "turbo", "scheb_two_factor", "ru2_f_two_factor""
```

**To Reproduce**
Steps to reproduce the behavior:
1. Take a running Part-DB-symfony install.  Update to or check out commit 8e6bc25d2e84beba314e073714d82a247936f3ed.
2. `composer install -o --no-dev --ignore-platform-req=ext-intl`
3. `yarn install`
4. `yarn build`
5. `php-8.1 bin/console doctrine:migrations:migrate`
6. `php-8.1 bin/console cache:clear`

**Expected behavior**
No error.

**Server Side**
- Part-DB Version: commit 8e6bc25d2e84beba314e073714d82a247936f3ed
- PHP Version: v8.1
- Database Server 8.0.28-0ubuntu0.20.04.3

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - Browser: Firefox
 - Version 102.0.1 (64-bit)

**Additional context**
If I subsequently load Part-DB-symfony in my browser I get an HTTP 500 error.  Nothing appears in my site's error.log file.  The only thing that appears in the access.log file is an error 500.

jbtronics, 2022-09-05:
Delete the `/config/packages/fos_ckeditor.yaml` file. It should not exist in new versions, as we don't use the package it belongs to anymore.

When updating you have to make sure no files from old versions are left over, or this can cause issues.

virtadpt, 2022-09-07:
I deleted the file and `php-8.1 bin/console cache:clear` got farther, but still looks like it errored out:

```
 // Clearing the cache for the prod environment with debug                     
 // false                                                                      

In DefinitionErrorExceptionPass.php line 54:
                                                                              
  Cannot autowire service "App\Command\UpdateExchangeRatesCommand": argument  
   "$base_current" of method "__construct()" is type-hinted "string", you sh  
  ould configure its value explicitly.                                        
                                                                              
cache:clear [--no-warmup] [--no-optional-warmers]
```

virtadpt, 2022-09-07:
As for making sure no files from old versions are laying around, that was something I've been meaning to ask about.  Is there a `yarn` function or a script someplace I haven't looked yet to clean out the old files?  Or should I be erasing the old install of Part-DB-symfony and re-synching a new one each time?

virtadpt, 2022-09-07:
Here's what I had to do: On my build machine, I had to erase the `vendor/` and `node_modules/` subdirectories before building the new version and uploading it to my host.  Now it seems to work.

closed 2022-09-17 by jbtronics