Part-DB/Part-DB-server #173: Error thrown while running command "cache:warmup".

virtadpt, 2022-09-05:
**Describe the bug**
When executing the command `php-8.1 bin/console cache:warmup` during a version upgrade the following error message is thrown:

```
15:27:15 CRITICAL  [console] Error thrown while running command "cache:warmup". Message: "Too few arguments to function App\Services\Attachments\AttachmentURLGenerator::__construct(), 5 passed in /home/virtadpt/merivis.virtadpt.net/Part-DB-symfony/var/cache/prod/ContainerKJhUyGL/getAttachmentURLGeneratorService.php on line 20 and exactly 6 expected" ["exception" => ArgumentCountError { …},"command" => "cache:warmup","message" => "Too few arguments to function App\Services\Attachments\AttachmentURLGenerator::__construct(), 5 passed in /home/virtadpt/merivis.virtadpt.net/Part-DB-symfony/var/cache/prod/ContainerKJhUyGL/getAttachmentURLGeneratorService.php on line 20 and exactly 6 expected"]

In AttachmentURLGenerator.php line 65:
                                                                              
  Too few arguments to function App\Services\Attachments\AttachmentURLGenera  
  tor::__construct(), 5 passed in /home/virtadpt/merivis.virtadpt.net/Part-D  
  B-symfony/var/cache/prod/ContainerKJhUyGL/getAttachmentURLGeneratorService  
  .php on line 20 and exactly 6 expected
```

**To Reproduce**
Steps to reproduce the behavior:
1. Take a running Part-DB-symfony install. Update to or check out commit 8e6bc25d2e84beba314e073714d82a247936f3ed.
2. `composer install -o --no-dev --ignore-platform-req=ext-intl`
3. `yarn install`
4. `yarn build`
5. `php-8.1 bin/console doctrine:migrations:migrate`
6. `php-8.1 bin/console cache:warmup`

**Expected behavior**
The Part-DB-symfony cache should be warmed up for regular use.

**Server Side**
- Part-DB Version: commit 8e6bc25d2e84beba314e073714d82a247936f3ed
- PHP Version: v8.1
- Database Server 8.0.28-0ubuntu0.20.04.3

**Desktop (please complete the following information):**
- OS: Arch Linux
- Browser: Firefox
- Version 102.0.1 (64-bit)

**Additional context**
If I subsequently load Part-DB-symfony in my browser I get an HTTP 500 error. Nothing appears in my site's error.log file. The only thing that appears in the access.log file is an error 500.

jbtronics, 2022-09-06:
Do you have cleared the cache beforehands? (Either by `php bin/console cache:clear` or by deleting the `var/cache` folder)?

Normally this should happen automatically during the `composer install` command, but maybe that does not finish correctly or your `--ignore-platform-reqs` somehow tampers with the container generation.

Besides that you should install the `ext-intl` extension for PHP if possible. I guess, the most calls are polyfilled by `symfony/intl` (even if you then can only use english formatting for datetimes and so), so it maybe work without. 
But this is not really tested, and this might cause issues.

virtadpt, 2022-09-07:
I tried (see https://github.com/Part-DB/Part-DB-symfony/issues/172).

The `composer install` bit is idiosyncratic to my setup: I'm on shared hosting, so I have to build Part-DB on another system and then `rsync` it up to my provider to finish the process.  I can't run composer there because it takes up too much CPU time and gets killed; I think the time limit is 60 seconds for a process that isn't a php-fpm on their servers.  For reasons I haven't had time to figure out, ext-intl for PHP on my build system either doesn't work or never gets detected.  It works fine on my hosting provider, though.

After clearing the cache and trying again, I get a new error:

```
 // Warming up the cache for the prod environment with debug
 // false

19:07:03 CRITICAL  [console] Error thrown while running command "cache:warmup". Message: "Too few arguments to function App\Services\Attachments\AttachmentURLGenerator::__construct(), 5 passed in /home/virtadpt/merivis.virtadpt.net/Part-DB-symfony/var/cache/prod/ContainerKJhUyGL/getAttachmentURLGeneratorService.php on line 20 and exactly 6 expected" ["exception" => ArgumentCountError { …},"command" => "cache:warmup","message" => "Too few arguments to function App\Services\Attachments\AttachmentURLGenerator::__construct(), 5 passed in /home/virtadpt/merivis.virtadpt.net/Part-DB-symfony/var/cache/prod/ContainerKJhUyGL/getAttachmentURLGeneratorService.php on line 20 and exactly 6 expected"]

In AttachmentURLGenerator.php line 65:

  Too few arguments to function App\Services\Attachments\AttachmentURLGenera  
  tor::__construct(), 5 passed in /home/virtadpt/merivis.virtadpt.net/Part-D  
  B-symfony/var/cache/prod/ContainerKJhUyGL/getAttachmentURLGeneratorService  
  .php on line 20 and exactly 6 expected                                      

cache:warmup [--no-optional-warmers]
```
Attempting to load my Part-DB-symfony install in a browser results in an HTTP 500 error.

virtadpt, 2022-09-07:
I also just realized that `Part-DB-symfony/var/log/*.log` are there and being used.  Is there something in there that would help?

virtadpt, 2022-09-07:
Here's what I had to do: On my build machine, I had to erase the `vendor/` and `node_modules/` subdirectories before building the new version and uploading it to my host.  Now it seems to work.

closed 2022-12-11 by jbtronics