Part-DB/Part-DB-server #177: Cross-compile docker image

deafloo, 2022-09-14:
**Is your feature request related to a problem? Please describe.**
I wanted to install PartDB on my Raspberry Pi 4 but the docker image provided is only for x86 architectures.

**Describe the solution you'd like**
You could use docker buildx to create images for several platforms

**Describe alternatives you've considered**
I tried to compile the image myself but I get a 'can't connect to database' error.

**Additional context**
A previous release a few months ago was working fine on my raspberry.

jbtronics, 2022-09-18:
In principle this totally works (and i have done this successfully in the past). The problem is that the cross building takes pretty long, which is a bit inconvenient for branches which receives changes often (as the build is often not finished until new commits are done)

jbtronics, 2023-02-06:
Now the images should also automatically be built for ARM64 and ARMv7, so the images should run on Raspberry Pi too.

closed 2023-02-06 by jbtronics