Part-DB/Part-DB-server #178: Edit or replace tags en masse.

virtadpt, 2022-09-15:
**Is your feature request related to a problem? Please describe.**
When I realize that I've misfiled or misspelled a tag name across my inventory (which happens when you clone entries to speed data entry) it takes a very long time to edit each and every item and correct the tag.

**Describe the solution you'd like**
It would be very helpful to have a function in Tools -> Tools where I could do a global search and replace on a tag (e.g., micocntrollers -> microcontrollers)

**Describe alternatives you've considered**
Search on the tag.  Open each hit in a separate tab.  Edit each hit in the separate tab to expose the tags.  Correct the tag.

**Additional context**
It would also be very helpful if the default component view could include the tags on a persistent basis.

jagauthier, 2022-09-19:
Can you drop into the database and just change it there?

virtadpt, 2022-09-19:
I could.  But isn't that killing a fly with a bazooka?

jagauthier, 2022-09-19:
I would consider bandaging a self-inflicted wound.

zanna-37, 2024-01-14:
I think that tags should be treated the same as categories, storage locations, etc.
It should be possible to edit, view and delete them in bulk.
One thing that I find frustrating is that when choosing the tags for a new part, I need to remember the tag name exactly, as the dropdown doesn't show them all.