Part-DB/Part-DB-server #181: Search returns all items

jagauthier, 2022-09-20:
Title pretty much sums it up!  

When executing a search all items are returned regardless of the search terms.

jagauthier, 2022-09-21:
I pulled in from master earlier this week:

```
commit 58ada496e4db0852889036f8f8658ce260756850 (HEAD -> master, origin/master, origin/HEAD)
Author: Jan Böhmer <mail@jan-boehmer.de>
Date:   Sun Sep 18 23:44:44 2022 +0200

    Fixed static analysis issues.

```

I am using MariaDB 10.8.3

jbtronics, 2022-09-23:
Can you pull the latest commit and try again (there are some issues in the version you have installed, which can trigger exceptions, too).
I could not replicate your described bug, in the newest version (and i tried on different systems)

jagauthier, 2022-09-23:
This is strange.  Pulled a new version:
```
commit 9c87202509ace9e520602277613df28ed485d1f8 (HEAD -> master, origin/master, origin/l10n_master, origin/HEAD)
Author: Jan Böhmer <mail@jan-boehmer.de>
Date:   Wed Sep 21 16:16:59 2022 +0200

    Allow more font sizes for editor on Label generator page

```

And took this video:

https://user-images.githubusercontent.com/7101432/191968665-a1af70f6-2572-4835-9835-edccc5fef372.mp4

jagauthier, 2022-09-23:
I think I have some weird local thing going on.  I changed banner.md and it did not change.
This might be an "outsmarted myself" situation.

jagauthier, 2022-09-23:
Yup. Big time on me.

closed 2022-09-23 by jagauthier