Part-DB/Part-DB-server #182: installation / launch issue

Bikoos, 2022-09-25:
**Describe the bug**
After the installation of the image on docker, i manage to connect to the web interface and i got this error : 

An exception occurred while executing a query: SQLSTATE[HY000]: General error: 1 no such table: users

**To Reproduce**
Steps to reproduce the behavior:
1. install image on synology docker
2. connect to web interface
3. issue

also tried to build the image on portainer
1. install image on container 
2. connect to web interface
3. same issue

**Screenshots**

[image]

**Server Side**

**php bin/console partdb:check-requirements**
return, all green : 

Checking PHP configuration...
=============================                                                                                                                      
 [OK] PHP version is sufficient.                                                                                        
 [OK] Opcache is enabled.                                                                                               
[OK] Opcache configuration is already performance optimized.                                                           
                                                                                                                        
Checking PHP extensions...
==========================
 [OK] PHP extension pdo_mysql is installed.                                                                             
[OK] PHP extension pdo_sqlite is installed.                                                                            
 [OK] PHP extension curl is installed.                                                                                  
[OK] PHP extension GD is installed.                                                                                    
[OK] GD has jpeg support.                                                                                              
 [OK] GD has png support.                                                                                               
[OK] GD has WebP support.      

**php bin/console partdb:version**
 [OK] Part-DB version: 1.0.0-alpha                                                                                      
 [INFO] PHP version: 8.1.10                                                                                             
 [INFO] Symfony version: 5.4.12                                                                                         
 [INFO] OS: Linux Part-DB 4.4.180+ #42661 SMP Mon Jun 27 15:05:48 CST 2022 x86_64                                       
 [INFO] PHP extension: Core, date, libxml, openssl, pcre, sqlite3, zlib, ctype, curl, dom, fileinfo, filter, ftp, hash, 
        iconv, json, mbstring, SPL, session, PDO, pdo_sqlite, standard, posix, readline, Reflection, Phar, SimpleXML,   
        tokenizer, xml, xmlreader, xmlwriter, mysqlnd, bcmath, gd, intl, pdo_mysql, sodium, xsl, zip, Zend OPcache  

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser chrome

**Additional context**
Synology DSM 7.1 
I have access to the terminal when i build the image with portainer.

Thank you for your dedication to this project.

jbtronics, 2022-09-25:
Have you run the database migrations? It looks that the basic database schemas are missing.

Bikoos, 2022-09-25:
Hello, 

Thank you for the tip/ fast reponse, I run this command : php bin/console doctrine:migrations:migrate 
and everything seems to work fine now, I will work on it this afternoon.

have a nice day

closed 2022-10-04 by jbtronics