Part-DB/Part-DB-server #189: Database migration failed, Duplicate key name attachment_types_idx_name.

virtadpt, 2022-10-15:
**Describe the bug**
While updating to the latest commit of Part-DB-symfony (6051c86521348f7f4c69d3f5262e028d15285030) I was attempting to run any outstanding database migrations and it failed immediately with the following error (slightly reformatted for legibility):

```
[notice] Migrating up to DoctrineMigrations\Version20221003212851
[error] Migration DoctrineMigrations\Version20220925162725 failed during Execution. Error:
    "An exception occurred while executing a query: SQLSTATE[42000]: Syntax error or
    access violation: 1061 Duplicate key name 'attachment_types_idx_name'"

17:00:36 CRITICAL  [console] Error thrown while running command
    "doctrine:migrations:migrate". Message: "An exception occurred while executing a query:
    SQLSTATE[42000]: Syntax error or access violation: 1061 Duplicate key name
    'attachment_types_idx_name'" ["exception" => Doctrine\DBAL\Exception\DriverException^
    { …},"command" => "doctrine:migrations:migrate","message" => "An exception occurred
    while executing a query: SQLSTATE[42000]: Syntax error or access violation: 1061
    Duplicate key name 'attachment_types_idx_name'"]
```

Note: I'm updating the install by moving the old directory out of the way and rsync'ing the latest build into its place, i.e., Part-DB-symfony/ was renamed to Part-DB-symfony.20221014/, and a new Part-DB-symfony/ was created next to it.

The .env.local file is unchanged.

**To Reproduce**
Steps to reproduce the behavior:
1. `composer update --ignore-platform-req=ext-intl`
2. `composer install -o --no-dev --ignore-platform-req=ext-intl`
3. On the host, `mv Part-DB-symfony Part-DB-symfony.yyyymmdd`
4. `yarn install`
5. `yarn build`
6. `rsync -avz Part-DB-symfony/ me@webserver:inventory.example.com/Part-DB-symfony/`
7. On the host, `php-8.1 bin/console doctrine:migrations:migrate`
8. :bomb: 

**Expected behavior**
The database migrations run in sequence.

**Server Side**
- Part-DB Version: Commit 6051c86521348f7f4c69d3f5262e028d15285030
- PHP Version: PHP 8.1.0 (cli) (built: Dec  9 2021 18:40:37) (NTS)
- Database Server Server version: 8.0.28-0ubuntu0.20.04.3 (Ubuntu)

**Additional context**
Add any other context about the problem here.

virtadpt, 2022-10-18:
Because the migrations didn't run, I was able to restore the previous version from backup, so this bug doesn't seem to damage older versions (provided that a backup is available).

virtadpt, 2022-11-13:
I just tried updating to commit ffceb5abc573a257efa3508ece8a2cf82149ff22.  Same problem.

virtadpt, 2022-11-29:
I used the command `php-8.1 bin/console doctrine:migrations:migrate --write-sql 20220925162725.sql DoctrineMigrations\\Version20220925162725  --dry-run` to dump all of the SQL statements that the migration executed and ran each one by hand.  Here are the SQL statements that failed when executed from database schema _DoctrineMigrations\Version20200502161750_ because they already existed:

* `CREATE INDEX attachment_types_idx_name ON attachment_types (name);`
* `CREATE INDEX attachment_types_idx_parent_name ON attachment_types (parent_id, name);`
* `ALTER TABLE attachments CHANGE type_id type_id INT NOT NULL;`
* `CREATE INDEX group_idx_parent_name ON `groups` (parent_id, name);`

virtadpt, 2022-11-29:
Following those manual changes, _DoctrineMigrations\Version20221003212851_ ran without problems.  _DoctrineMigrations\\Version20221114193325_ also executed without trouble.

jbtronics, 2022-11-29:
I can not find where these indices would be created before, so I dont think that this is really an issue in Part-DB. The migrations are automatically applied on an empty database, and I encountered no issue in that aspect, the same is for the databases I maintain.

So I would guess, that this problem was somehow caused by a migration which only partly finished at your database, and which has already created the indices, but did not finished. When you then re run the migration, it will most likely fail because, the indices are already existing.

There is no real good solution to the problem with partly failed migrations on MySQL as it does not support transaction on schema changes. So when the database schema was somehow changed and then something else during the migration fails, Part-DB can not really revert the changes (and you have to restore a database backup manually).

virtadpt, 2022-12-05:
Hmm.  That is entirely likely.  I guess I'll write this up as an article for MySQL newbies (which I'm only slightly more than).

closed 2022-12-05 by virtadpt