Part-DB/Part-DB-server #194: Migration error while upgrading to latest (75cf3dc)

elperepat, 2022-11-28:
**Describe the bug**
Update to 75cf3dc fails to migrate properly from c6ad14d

**To Reproduce**
Steps to reproduce the behavior:
cd /var/www/partdb
git pull
composer install -o --no-dev
yarn install
yarn build
php bin/console cache:warmup
php bin/console doctrine:migrations:migrate

**Expected behavior**
Migrate without issue

**Server Side**
- Part-DB Version: [e.g. Part-DB 1.0 75cf3dc
- PHP Version: [e.g. PHP 7.4.3]
- Database Server [e.g. MySQL 8.0.31]

**Additional context**

Here is the log from the migration failing:

 WARNING! You are about to execute a migration in database "partdb" that could result in schema changes and data loss. Are you sure you wish to continue? (yes/no) [yes]:
 > 

[notice] Migrating up to DoctrineMigrations\Version20221114193325
[warning] 
!!! All permissions were reset! Please change them to the desired state, immediately !!!
[warning] 
!!! For security reasons all users (except the admin user) were disabled. Login with admin user and reenable other users after checking their permissions !!!
[warning] 
!!! For more infos see: https://github.com/Part-DB/Part-DB-symfony/discussions/193 !!!
[error] Migration DoctrineMigrations\Version20221114193325 failed during Execution. Error: "An exception occurred while executing a query: SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'groups ADD permissions_data LONGTEXT NOT NULL COMMENT '(DC2Type:json)', DROP per' at line 1"
20:00:18 CRITICAL  [console] Error thrown while running command "doctrine:migrations:migrate". Message: "An exception occurred while executing a query: SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'groups ADD permissions_data LONGTEXT NOT NULL COMMENT '(DC2Type:json)', DROP per' at line 1" ["exception" => Doctrine\DBAL\Exception\SyntaxErrorException^ { …},"command" => "doctrine:migrations:migrate","message" => "An exception occurred while executing a query: SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'groups ADD permissions_data LONGTEXT NOT NULL COMMENT '(DC2Type:json)', DROP per' at line 1"]

In ExceptionConverter.php line 86:
                                                                                                                                  
  An exception occurred while executing a query: SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in yo  
  ur SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'groups ADD per  
  missions_data LONGTEXT NOT NULL COMMENT '(DC2Type:json)', DROP per' at line 1                                                   
                                                                                                                                  

In Exception.php line 28:
                                                                                                                                  
  SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in your SQL syntax; check the manual that correspond  
  s to your MySQL server version for the right syntax to use near 'groups ADD permissions_data LONGTEXT NOT NULL COMMENT '(DC2Ty  
  pe:json)', DROP per' at line 1                                                                                                  
                                                                                                                                  

In Connection.php line 69:
                                                                                                                                  
  SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in your SQL syntax; check the manual that correspond  
  s to your MySQL server version for the right syntax to use near 'groups ADD permissions_data LONGTEXT NOT NULL COMMENT '(DC2Ty  
  pe:json)', DROP per' at line 1                                                                                                  
                                                                                                                                  

doctrine:migrations:migrate [--write-sql [WRITE-SQL]] [--dry-run] [--query-time] [--allow-no-migration] [--all-or-nothing [ALL-OR-NOTHING]] [--configuration CONFIGURATION] [--em EM] [--conn CONN] [--] [<version>]

jbtronics, 2022-11-28:
Groups and (in newer versions) Users are reserved keywords in MySQL 8 (but not in MariaDB), so the table names must be escaped properly in migrations to work with MySQL 8.

It should now work.

closed 2022-11-28 by jbtronics

elperepat, 2022-11-28:
It's working properly now, thanks!