Part-DB/Part-DB-server #198: Add Supplier part number in Purchase information

elperepat, 2022-12-12:
**Is your feature request related to a problem? Please describe.**
When dealing with suppliers, they often have their own part number, different from manufacturer part number.

**Describe the solution you'd like**
A new field named "Supplier part number" added to the "Purchase information" tab for each supplier  would solve this issue.

**Describe alternatives you've considered**
Adding that information to comment, but wouldn't be as clear as a new field when dealing with multiple suppliers for a part as each have their own part number for the same product.

**Additional context**
See example for a simple fuse and different suppliers in screenshot.

[image]
[image]

jbtronics, 2022-12-12:
I would say, that this is the "order number" field in the priceinformation tab, or is there any reason that you can not use it for this?

[image]

elperepat, 2022-12-13:
I was actually tracking the order number (invoice number of the actual order) in there. But yeah, I guess it could serve that purpose instead.

But I see it now, the value entered here is listed in the column "Partnr." when I show the "Shopping information" tab when not in edit mode. There would be a cleanup to do in this area. When editing, the label is "Order number". Maybe it should be changed to "Supplier part number". Same thing with the supplier: the green button in edit mode is "Add distributor". I think it should be "Add supplier".

By the way, thanks for developing and maintaining partDB. There really isn't anything quite like it and it's much needed.

Pat

jbtronics, 2022-12-13:
Yeah maybe the labeling in english is a bit misleading, I can change that.

As you can order a part multiple times from the same supplier, and invoice numbers are not really part related, an invoice number would be more interesting for an (not yet existing) order system.

jbtronics, 2023-02-06:
Fixed now.

closed 2023-02-06 by jbtronics