Part-DB/Part-DB-server #201: How to change the amount of an item?

Dschadu, 2022-12-27:
Hi,

sorry if I'm missing it, but how do I change the amount of an item? Only way I found is:
Click the part -> Edit (the Button behind the Name) -> Stocks -> Change the Number manually. 

There has to be an easier way, like with the old PartDB?
https://www.mikrocontroller.net/wikifiles/4/44/Parts_info.png

What am I missing? Thanks in advance!

jbtronics, 2022-12-30:
Currently editing the instock via the edit page is the only possibility. 
But i will add the feature that you can easily add or subtract (or move) instock from the part info page. 

This is a bit more difficult than in the older Part-DB versions, as now a part can have stocks at multiple locations.

jbtronics, 2023-01-08:
In the latest version there is now a modal for easy withdrawal/addition of parts from a lot, which can be opened from the lots info page.

[image]

closed 2023-01-08 by jbtronics

elperepat, 2023-01-09:
Could the modal form and the fields be a little bit narrower, to fit on a small screen? As it is now, I have to scroll sideways on my cellphone to view the form correctly. The rest of the website is liquid, as expected. This form isn't.

[image]

Thanks!

jbtronics, 2023-01-10:
The modal should now properly fit the screen.