Part-DB/Part-DB-server #202: BOM Check

iyzpx, 2023-01-03:
**Is your feature request related to a problem? Please describe.**
When dealing with many boards and components, checking all parts are present is sometimes very frustrating. Printing and checking by hand isn't up to date and also a waste of paper. With an existing datebase like partdb, this could be automated.

**Describe the solution you'd like**
A possibility to upload a BOM (csv format) and a check of each components with the existing database. Since each BOM is differently structured, a selection field is probably required to select the column for components and quantity. Also a multiplier, if more than one board has to be assembled.

The parts may be highlighted like this:
Available
Not sufficient (or only for x boards)
Not found
Multiple parts found

**Describe alternatives you've considered**
Since there is no API for the PartDB as far as I know, I could write a simple php script that accesses the database.

jbtronics, 2023-01-23:
This should now be (more or less) possible using the project and build feature of Part-DB.
You can define your BOM as a project and then it will be show, how often you can build this project with the existing components. When trying to build the project a certain times, it will be shown what parts are missing and how often. (see screenshots below)
When enough parts are stocked, it allows you to withdraw the correct amount of parts from stock needed for the build, all at once.

[image]
[image]

However it is not possible yet, to import BOMs from CSVs or similar yet, so I will leave this issue open.

iyzpx, 2023-01-24:
@jbtronics Yeah, I immediately installed this version after I saw your commits. It already works great and is very helpful!

Would it make sense to implement some kind of part reservation before finally building a project? I am thinking about building several projects that contains the same parts. So they could be reserved first, and when the project is assembled, the reserved parts can checked out.

I guess this might give a better overview when planning multiple projects. I would be interested in your thoughts about this.

jbtronics, 2023-03-18:
BOM entry import is now possible (for KiCAD Pcbnew BOM files, or you have to change the field names manually), so I will close this issue

closed 2023-03-18 by jbtronics