Part-DB/Part-DB-server #203: Create categories, storage locations etc. at part creation

suuppl, 2023-01-25:
**Is your feature request related to a problem? Please describe.**
Not really a problem, just quality of life.

**Describe the solution you'd like**
I would like to create categories, storage locations, packages etc. at part creation.
My ideal solution:
Start typing in the category search field. If the category exists, i can just use it. 
If not, part-db should create it. For subcategories, use slashes.

Example:
I am creating a part "BC817-25", there is a category "Transistors" and a subcategory "Transistors/BJT" but no "Transistor/BJT/NPN".
So I type "Transistor/BJT/NPN" in the category search field. As this category doesn't exist yet, show a button "Create new category"

**Describe alternatives you've considered**
Create all categories (storage locations, ...) before filling in parts or go back and forth between categories and parts, but this slows down the part creation considerably

**Additional context**
Thank you for your work!

virtadpt, 2023-01-25:
It would be handy.

jbtronics, 2023-01-30:
In the newest version this is now possible. You can type in the new name in the entity select field, and click "Add".
Using "->" you can create nested structures.

[image]

closed 2023-01-30 by jbtronics

suuppl, 2023-01-31:
That's awesome, thank you!